aurora-opensource/xviz #349: Localhost says upgrade required

DeepTecher, 2019-02-25:
Thanks for sharing the very useful tool for AV visualization.  

I think I managed to run the example successfully because “Checking for index file ,Loaded 154 frames” has already displayed on the terminal.However,the localhost says upgrade required.

Can anyone here help me solve this problem?

xintongxia, 2019-02-25:
Could you provide the stack trace of your error?

DeepTecher, 2019-02-25:
After run the demo script`./scripts/run-kitti-example.sh` ,terminal shows as following：
>XVIZ Server started.
Ctrl-c to terminate.
Loading frames from  path/xviz/scripts/../data/generated/kitti/2011_09_26/2011_09_26_drive_0005_sync/
Checking for files:  path/xviz/data/generated/kitti/2011_09_26/2011_09_26_drive_0005_sync/1-frame.glb
Checking for index file
Loaded 154 frames

However, http://localhost:8081  does not display anything other than showing  **upgrade required**

xintongxia, 2019-02-25:
Our `XVIZ` server is using `WebSocket` to communicate with client (browser), you can check [`streetscape.gl`](https://github.com/uber/streetscape.gl), and it has a simple [example](https://github.com/uber/streetscape.gl/tree/master/examples/get-started) which can consume the data from `XVIZ` server.

After you install all the dependencies and get the `XVIZ` server up and running like you did previously, you can simply start the `streetscape.gl` example by doing `yarn start-streaming-local`.

But we do need improve our documentation.

DeepTecher, 2019-02-25:
Thanks.Now everything is OK.

closed 2019-02-25 by DeepTecher

KevinYuk, 2019-03-29:
Hi @DeepTecher,

Could you please explain how do you solve the issue above?

Thanks a lot.

KevinYuk, 2019-03-29:
Hi @DeepTecher ,

I also can run it by using yarn start-streaming-local from @xintongxia . Thanks a lot.

However, I found that xviz server is started on localhost:8081, and xviz client (which is lauched by "start-streaming-local") connects to xviz server on localhost:8080.
@xintongxia Could you please explain it?

Many thanks.

Avkash, 2019-09-20:
Hi there, I am having the exact same problem as @DeepTecher & @KevinYuk with "Upgrade required" error on browser.

$ ./scripts/run-kitti-example.sh
##
## XVIZ Server started.
## Ctrl-c to terminate.
##
xvizserver-log: Listening on port 8081

>> At this point the browser returns "Upgrade Required"...

Note: I can also run streetscape.gl > examples/getstart with "yarn start" successfully however not able  to run the "xviz server". 
Any idea?