aurora-opensource/xviz #350: Running straight scenario with --loop errors

gerardsimons, 2019-02-25:
Hey guys,

I am trying to implement a streaming XVIZ server of my own. The scenarios seems to be very helpful for this. Unfortunately when i try it with looping as such:

`node index.js --loop --scenario straight`

I get an error:

```
Loaded 300 frames
Unpacking 300 frames into memory
WARNING: for long logs you might not have enough memory
/Users/gerard/Git/xviz/examples/server/xviz-serve-data.js:682
    throw new Error('Unknown frame type');
    ^
```

indeed if I inspect the frames coming in, they are `undefined`

xintongxia, 2019-02-25:
After digging into the code, I found out this is related to the `--loop` flag in the script. Before I have more context about this fix, you could try to disable `--loop` for the time being. 

Will have a follow-up fix soon.

jlisee, 2019-02-25:
#352 will disable using both of these features and provide a clear error message.   Once that lands this will become a feature request to enable looping with scenarios :)

jlisee, 2019-02-25:
We have landed #352 so you'll get a clear error message that this is not currently possible, we can look into support this in the future.

twojtasz, 2019-02-27:
I haven't seen this mentioned but `--duration <seconds>` with a high # of seconds should work to get data flowing