aurora-opensource/xviz #355: How do I use the CLI tools?

gerardsimons, 2019-02-26:
I am currently trying to develop my own live streaming XVIZ server. It would be nice to have a few example of the CLI tool. I think these would be helpful for debugging, but I have no idea how to run them. Some documentation on this and what each tool is good for would be nice.

jlisee, 2019-02-26:
Thanks for filing an issue about this.  I have some notes on plan and the existing usage of the tool and I can get that as a start of the documentation for the tool itself.   I am hoping to get to it by the end of the week but here are some quick notes:

You should be able to run the tool with JS tool chain, ie. `node` and `npm`/`yarn` so install like so (I have just run these commands and they don't work, I am looking into it and making it work see: ):

```
yarn add @xviz/cli
npx xviz
```

```
xviz <command> [<args>]

The xviz tool support the following commands for exploring
and saving XVIZ data.

   dump       Output XVIZ content to stdout
  validate   Ensure data and server conform to the specification
```

Dump the data from the server (all of it) in condensed form:
```
xviz dump -c ws://localhost:8085 630e522f-d2b1-403c-a9a3-468b398cbf60
```

Dump EVERYTHING:
```
xviz dump ws://localhost:8085 630e522f-d2b1-403c-a9a3-468b398cbf60
```

Validate what you are producing with a summary of errors at the end:

```
xviz validate -c ws://localhost:8085 630e522f-d2b1-403c-a9a3-468b398cbf60
```

See all the validation errors (note, these are a little verbose but they point you in the direction):
```
xviz validate ws://localhost:8085 630e522f-d2b1-403c-a9a3-468b398cbf60
```

jlisee, 2019-03-13:
I have landed an initial set of documented in #376, you can see it in master here: https://github.com/uber/xviz/tree/master/docs/xviz-tool

closed 2019-03-13 by gerardsimons