aurora-opensource/xviz #358: how to convert and start Demo application

pgh41, 2019-02-27:
In your homepage at getting-started page to convert KITTI to XVIZ Fromat
write like this
[image]
when i write yarn start -d 2011_09_26/2011_09_26_drive_0005_sync
error massage that have to use -o option.
how to solve this problem?
and at the github page ./scripts/download-kitti-data.sh
and  ./scripts/run-kitti-example.sh also doesn't run
how to i solve this problem.

pgh41, 2019-02-28:
> In your homepage at getting-started page to convert KITTI to XVIZ Fromat
> write like this
> [image]
> when i write yarn start -d 2011_09_26/2011_09_26_drive_0005_sync
> error massage that have to use -o [...]

closed 2019-02-28 by pgh41

RunOrVeith, 2019-03-04:
For anyone wondering at this, the following things in that example appear to be outdated:

- Relative paths are not resolved as described, but based on the directory where the script is started
- I had to run `yarn add @xviz/builder` after running `yarn` because that was not installed automatically
- The `-o` option is now required. You need to specify an output folder. The expected one seems to be `data/examples/kitti/generated/YOUR_DATASET_NAME`

Mandy3253, 2019-03-17:
i also meet a problem when i convert  example data according to the [quick start demo](https://github.com/uber/xviz),
[image]
when i write  ./scripts/download-kitti-data.sh, it display a error
[image]
i look the file line40 but  i don't kown what mean of this and how to solve this problem
[image]
would someone can help me?

RunOrVeith, 2019-03-17:
You don't have wget installed. That is completely unrelated. Use 'sudo apt
install wget' to install it on Linux.

Mandy3253, 2019-03-17:
> You don't have wget installed. That is completely unrelated. Use 'sudo apt install wget' to install it on Linux.
> […](#)

thank you so much,but  i ran it on windows .how can i solve it? or  this project  only can run  on Linux?

Mandy3253, 2019-03-18:
> You don't have wget installed. That is completely unrelated. Use 'sudo apt install wget' to install it on Linux.
> […](#)
thank you very much.because  i didn't use Linux rather ran it on windows.

MatthiasThDs, 2019-05-13:
Hello,
i am on windows too. I wrote the necessary commands to download and run the script myself. but i get the following error:

```
C:\Projects\xviz\examples\converters\kitti>yarn
yarn install v1.15.2
warning package.json: No license field
info No lockfile found.
warning jskitti@1.1.0: No license field
[1/4] Resolving packages...
warning xml2json > hoek@4.2.1: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
warning xml2json > joi@13.7.0: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
warning xml2json > joi > hoek@5.0.4: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
warning xml2json > joi > topo@3.0.3: This module has moved and is now available at @hapi/topo. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.
warning xml2json > joi > topo > hoek@6.1.3: This module has moved and is now available at @hapi/hoek. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.
[2/4] Fetching packages...
info fsevents@1.2.9: The platform "win32" is incompatible with this module.
info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
[4/4] Building fresh packages...
[-/2] ⠄ waiting...
error C:\Projects\xviz\examples\converters\kitti\node_modules\node-expat: Command failed.
Exit code: 1
Command: node-gyp rebuild
Arguments:
Directory: C:\Projects\xviz\examples\converters\kitti\node_modules\node-expat
Output:
C:\Projects\xviz\examples\converters\kitti\node_modules\node-expat>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "" rebuild )
gyp info it worked if it ends with ok
gyp info using node-gyp@3.8.0
gyp info using node@10.15.3 | win32 | x64
gyp http GET https://nodejs.org/download/release/v10.15.3/node-v10.15.3-headers.tar.gz
gyp http 200 https://nodejs.org/download/release/v10.15.3/node-v10.15.3-headers.tar.gz
gyp http GET https://nodejs.org/download/release/v10.15.3/SHASUMS256.txt
gyp http GET https://nodejs.org/download/release/v10.15.3/win-x86/node.lib
gyp http GET https://nodejs.org/download/release/v10.15.3/win-x64/node.lib
gyp http 200 https://nodejs.org/download/release/v10.15.3/SHASUMS256.txt
gyp http 200 https://nodejs.org/download/release/v10.15.3/win-x86/node.lib
gyp http 200 https://nodejs.org/download/release/v10.15.3/win-x64/node.lib
gyp info spawn C:\Program Files (x86)\Python27\python.exe
gyp info spawn args [ 'C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-G',
gyp info spawn args   'msvs_version=auto',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Projects\\xviz\\examples\\converters\\kitti\\node_modules\\node-expat\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\MatthiasTh\\.node-gyp\\10.15.3\\include\\node\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\MatthiasTh\\.node-gyp\\10.15.3',
gyp info spawn args   '-Dnode_gyp_dir=C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=C:\\Users\\MatthiasTh\\.node-gyp\\10.15.3\\<(target_arch)\\node.lib',
gyp info spawn args   '-Dmodule_root_dir=C:\\Projects\\xviz\\examples\\converters\\kitti\\node_modules\\node-expat',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\Projects\\xviz\\examples\\converters\\kitti\\node_modules\\node-expat\\build',
gyp info spawn args   '-Goutput_dir=.' ]
Warning: Missing input files:
C:\Projects\xviz\examples\converters\kitti\node_modules\node-expat\build\deps\libexpat\..\..\..\deps\libexpat\version.c
Warning: unrecognized setting VCCLCompilerTool/MultiProcessorCompilation
Warning: unrecognized setting VCCLCompilerTool/MultiProcessorCompilation
Warning: unrecognized setting VCCLCompilerTool/MultiProcessorCompilation
Warning: unrecognized setting VCCLCompilerTool/MultiProcessorCompilation
Warning: unrecognized setting VCCLCompilerTool/MultiProcessorCompilation
Warning: unrecognized setting VCCLCompilerTool/MultiProcessorCompilation
gyp info spawn C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe
[... cut ...]
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Projects\xviz\examples\converters\kitti\node_modules\node-expat
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
```