aurora-opensource/xviz #384: Remove text-encoding dependency

shanshanzhu, 2019-03-02:
Issues:

The repo use `text-encoding` lib in the utils https://github.com/uber/xviz/blob/master/modules/parser/src/utils/text-encoding.js#L18

`text-encoding` is deprecated and no longer maintained. Nodejs expose TextEncoder from `util` module in v8, and from `global` module in v11.
https://nodejs.org/api/util.html#util_class_util_textencoder
https://www.npmjs.com/package/text-encoding

Suggested fix:
1. Completely removing 'text-encoding` dependency;
2. Should we replace it with TextEncoder from nodejs? (is uber/xviz meant to function in node env?)

ibgreen, 2019-03-02:
This might also be a question for loaders.gl, which currently includes this polyfill under node (and xviz uses loaders.gl). 

Some vis.gl framework based apps support IE11, I assume that they probably still need this. We could update the instructions for building for IE11 to mention including this polyfill, and not include it by default. @tsherif

tsherif, 2019-03-02:
It looks like TextEncoder isn't supported in IE or Edge: https://caniuse.com/#feat=textencoder

(Edge surprises me, actually. I'll double check that on my Windows machine at home.)

Pessimistress, 2019-03-02:
We have 

```
  "browser": {
    "text-encoding": false
  }
```

in package.json so it wouldn't work in these browsers regardless.

tsherif, 2019-03-02:
I can confirm that TextEncoder is not supported in Edge.