aurora-opensource/xviz #386: Single file static XVIZ Log

hfchendrikx, 2019-03-02:
Hi, First of all, this is a great project and I would really like to make use of it, thanks for making it open source.

I was wondering if it is possible to pack all the single frame files into one large file? Which then has some kind of header at the front so that the client can skip to the correct frame using range headers, like a video stream. Is this implemented already or are you planning on implementing this?

jlisee, 2019-03-02:
We do not currently have this but we had some thoughts around it.  Can you describe how you want to use these files?

The basic idea was to move our multi-file on disk format into a "glb" like format where chunk 0 is a JSON index, chunk 1 is the metadata, and the rest of the chunks are the frame/updates.  So a client can read the front of the file then any chunk they want based on the time range they want to see.  We could also update the index to include which streams are in which chunk, such you can further focus reading based on the chunks.

twojtasz, 2019-03-06:
I see the this single file as serving a few purposes
1. Convenience 
2. Streaming
3. Optimization

Each case leads to a different internal structure.  I looked over at glTF related issues on streaming and it is on the radar, but not really a target for the format.  The scope for general glTF is different and our usage models the video stream closely, so it would be handy for an HTTP server.

In general 1 & 2 align well for our case and is definitely worth doing.

https://github.com/KhronosGroup/glTF/issues/1238
https://gist.github.com/javagl/bfde5cfab4240843120ed6eb38f4af87

hfchendrikx, 2019-03-10:
Thanks for the answers! 

My usecase would be converting logs offline from the rosbag format to XVIZ and then storing them in some http accessible storage, like S3. This would then not require any special server. Of course this would also be possible with all the single frame files, but in my opinion it's easier to manage and cleaner if there is only one file per log.

MikeChou, 2020-06-23:
My solution:
1. gzip each frame.glb, store its sizes before & after compression
2. concatenate the gzipped files orderly into a single large file, store it in S3 (setting Content-Encoding: gzip header)
3. put the two sizes of all frames into the meta file, or a separate index file
4. write a custom XVIZFileLoader using the index info in meta file to calculate the offsets (both before & after compression), and use http Range header to fetch part of the large file or you can download/stream the whole file from beginning to end. (use the sizes after compression to calculate the offsets for the http request range, use the sizes before compression to calculate the offsets for frame data in the decompressed result. gzip decompression is handled automatically by the browser)

Using gzip compression makes the result file significantly smaller. The key point is that the large file concatenated from individually compressed gzip files is still valid as long as you know the offsets to split it into what you want.

twojtasz, 2020-06-23:
@MikeChou this would be a great example to have added or at least we could like to any project you have that implemented this