aurora-opensource/xviz #387: Unable to replicate playground code in ws server (INCOMPLETE error)

gerardsimons, 2019-03-02:
Hi,

I am building a very simple streaming XVIZ server. I think I have the simplest one, and thought it'd be good to start with the examples you see in playground. In general I really enjoy this workflow of trying something in the playground (really cool example!) and then putting it in my application. 

Unfortunately, trying to run this WS server:

```
const fs = require('fs');
const WebSocket = require('ws');
const path = require('path');
const process = require('process');

const xviz = require('@xviz/builder');
const XVIZBuilder = xviz.XVIZBuilder;
const XVIZMetadataBuilder = xviz.XVIZMetadataBuilder;
const {deltaTimeMs} = require('./serve');

console.log("Simple XVIZ server");

// Connection State
class ConnectionContext {
    constructor(settings, metadata, allFrameData, loadFrameData) {
        this.onConnection.bind(this);
        this.onClose.bind(this);
        this.onMessage.bind(this);
    }

    log(msg) {
        console.log("CC", msg)
    }

    logMsgSent(send_time, index, real_index, tag, ts = 0) {
        const t_from_start_ms = deltaTimeMs(this.t_start_time);
        const t_msg_send_time_ms = deltaTimeMs(send_time);
        this.log(
          ` < Frame(${tag}) ts:${ts} ${index}:${real_index} in self: ${t_msg_send_time_ms}ms start: ${t_from_start_ms}ms`
        );
      }

    sendMetadata() {

        // metadata
        var xvizMetadataBuilder = new XVIZMetadataBuilder();

        // Exact copy of playground
        xvizMetadataBuilder
        .stream('/vehicle_pose')
        .category('pose')

        .stream('/object/tracking_point')
        .category('primitive')
        .type('circle')
        .coordinate('VEHICLE_RELATIVE')
        .streamStyle({
          fill_color: '#fb0'
        })

        .startTime(1000)
        .endTime(1010);

        var metaData = JSON.stringify(xvizMetadataBuilder.getMetadata());
        console.log('--META--');
        console.log(metaData);

        this.ws.send(metaData, {compress: true});
        const frame_send_time = process.hrtime();
        this.logMsgSent(frame_send_time, 1, 1, 'metadata');
    }

    sendFrame() {
        const envelope = {
            type: `xviz/${type}`,
            data: msg
          };
          const data = JSON.stringify(envelope);
          this.ws.send(data, options, callback);
    }
  
    onConnection(ws) {
        this.log('> Connection from Client.');

        this.t_start_time = process.hrtime();
        this.ws = ws;
    
        // Respond to control messages from the browser
        ws.on('message', msg => this.onMessage(msg));
    
        // On connection send metadata
        this.sendMetadata();
    
        // 'live' mode will not get the 'xviz/transform_log' message
        // so start sending immediately
        // this._sendTestFrames();
    }
  
    onClose(event) {
      this.log(`> Connection Closed. Code: ${event.code} Reason: ${event.reason}`);
    }
  
    onMessage(message) {
      const msg = JSON.parse(message);
  
      this.log(`> Message ${msg.type} from Client`);
    }
}

function setupWebSocketHandling(wss) {
    // Setups initial connection state
    wss.on('connection', ws => {

        console.log("WSS on connection ... ")
        const context = new ConnectionContext();
        context.onConnection(ws);
    });

    console.log("Setup WebSocket handling");
}

// Try send some meta data
const wss = new WebSocket.Server({port: 8081});
setupWebSocketHandling(wss)

```

with the streetscape.gl get-started (yarn start-streaming) results in an "INCOMPLETE" error so it doesnt work. Any ideas?

gerardsimons, 2019-03-02:
In fact, it looks like even the generated JSON in the playground is not valid? To make it valid I have to wrap the version in a "data" tag. So not 

```
"type": 'xviz/metadata',
"version": "2.0.0",
"streams": { ... }
```

But rather:
```
"type": 'xviz/metadata',
  "data" :  {
    "version": "2.0.0",
  },
  "streams": { ... }
```

I also noticed the XVIZPlayground uses an XVIZWriter to turn the resulting metadata in a message. Do I need to do that? Is a simple `JSON.stringify(xvizMetadataBuilder.getMetadata())` not sufficient?

jlisee, 2019-03-02:
## Issue

When running over a web socket we put XVIZ data into an envelop object with just `type` and `data` fields. Where `type` is `xviz/TYPE` and `data` is the actual content returned by the builders.  This allows you to send application specific data on the web socket as well as XVIZ data.  You can read more about it [here in the docs](https://avs.auto/#/xviz/protocol/protocol-implementations/json-protocol?section=websocket-data-envelope).

A concrete example is your metadata should look like:

```json
{
    "type": "xviz/metadata",
    "data": {
        "version": "2.0.0"
    }
}
```

## Solution

Make your send method look like this:
```js
    var envelopedMsg = {
      type: 'xviz/metadata',
      data: xvizMetadataBuilder.getMetadata()
    };
    var metaData = JSON.stringify(envelopedMsg);
```

Then you'll see (I have to ctrl-c) the server
```
$ npx xviz dump -c ws://localhost:8081 logname
[CONNECTED]
[< START] 
[> METADATA] version: 2.0.0
[< TRANSFORM_LOG] LOG-START - LOG-END (tid: f8b38a41-59fa-44b9-9311-cd612886bb37)
^CClosing
[CONNECTION CLOSED]
```

## Server Dev Tip
I have found the best way to test a server is to use the command line client we have built.  You need to grab it from master:

```
# make sure you have yarn and node installed
git clone git@github.com:uber/xviz.git
cd xviz
./script/bootstrap.sh
```

Then you can test queries against your client:

```
$ npx xviz dump -c ws://localhost:8081 logname 
[CONNECTED]
[< START] 
Unknown message format { version: '2.0.0',
  streams: 
   { '/vehicle_pose': { category: 'pose' },
     '/object/tracking_point': 
...
```

Also @twojtasz is working on morphing our example code into a real server module.

gerardsimons, 2019-03-03:
Ah that's great advice and fixes my issue so far. Thanks for taking the extra effort in giving me some dev tips too.

closed 2019-03-03 by gerardsimons