aurora-opensource/xviz #388: Unable to install @xviz/builder

gerardsimons, 2019-03-02:
Starting from scratch by trying to install @xviz-builder in a new directory 

`npm install @xviz/builder`

results in an error:

```
npm WARN deprecated text-encoding@0.6.4: no longer maintained
npm WARN enoent ENOENT: no such file or directory, open '/Users/gerard/Desktop/xviz-test/package.json'
npm WARN xviz-test No description
npm WARN xviz-test No repository field.
npm WARN xviz-test No README data
npm WARN xviz-test No license field.

npm ERR! path /Users/gerard/Desktop/xviz-test/node_modules/@loaders.gl/gltf/bin/glbdump.js
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall chmod
npm ERR! enoent ENOENT: no such file or directory, chmod '/Users/gerard/Desktop/xviz-test/node_modules/@loaders.gl/gltf/bin/glbdump.js'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/gerard/.npm/_logs/2019-03-02T16_14_27_709Z-debug.log
```

I have attached the log file.
[2019-03-02T16_14_27_709Z-debug.log](https://github.com/uber/xviz/files/2922270/2019-03-02T16_14_27_709Z-debug.log)

My npm version is 6.7.0

jlisee, 2019-03-02:
I get the same thing with npm version 6.4.1, as a work around you can use `yarn`:

```
$ yarn add @xviz/builder
yarn add v1.10.1
info No lockfile found.
[1/4] 🔍  Resolving packages...
...
└─ two-sum@1.0.0
✨  Done in 15.25s.
$ ls node_modules/@xviz/builder
README.md    dist         node_modules package.json
```

Pessimistress, 2019-03-03:
@ibgreen

ibgreen, 2019-03-04:
This was fixed in https://github.com/uber-web/loaders.gl/pull/103 and the latest published version of loaders.gl should have the fix. Try bumping the xviz dependency and let me know.

gerardsimons, 2019-03-04:
Not sure how I bump the version from commandline `npm install @xviz/builder` ? What version do I specify?

twojtasz, 2019-03-05:
@gerardsimons I will update the package inside *@xviz/builder* and then we will see about publishing a new one.  In the meantime I hope the yarn work around works?

gerardsimons, 2019-03-06:
Indeed it does, and a colleague found a workaround for NPM too by changing the loader dependency, which is probably what you meant (sorry I am a big Node noob). Thanks again.

ziyadk1997, 2019-06-01:
im having the same problem can someone elaborate more on how to fix this problem

twojtasz, 2019-06-01:
@ziyadk1997 I just ran with node v10.15.3  yarn 1.10  the following on mac and it appears to have worked.
```
$ mkdir tmp  && cd tmp
$ yarn init
$ yarn add @xviz/builder@^1.0.0-beta.14
$ cat package.json
{
  "name": "test",
  "version": "1.0.0",
  "main": "index.js",
  "license": "MIT",
  "dependencies": {
    "@xviz/builder": "1.0.0-beta.14"
  }
}
```

twojtasz, 2019-06-01:
let me know  how your setup is different.