aurora-opensource/xviz #391: KITTI example: base-converter.js file selection must be smarter

RunOrVeith, 2019-03-04:
I just tried running the KITTI example on a different piece of KITTI: 2011_09_26_drive_0009
For downloading I ran
`scripts/download-kitti-data.sh 2011_09_26_drive_0009`

Then I ran the converter in examples/converters/kitti:

`yarn start -d ../../../data/kitti/2011_09_26/2011_09_26_drive_0009_sync -o ../../../data/generated/kitti/2011_09_26/`
Running the converter crashes after displaying
`processing gps data frame 443/447`

with the following stacktrace:

```
(node:28813) UnhandledPromiseRejectionWarning: TypeError: Path must be a string. Received undefined
    at assertPath (path.js:28:11)
    at Object.join (path.js:1236:7)
    at LidarConverter.join (/home/veith/understandai/avs-demo/xviz/examples/converters/kitti/src/converters/base-converter.js:41:30)
    at tryCatch (/home/veith/understandai/avs-demo/xviz/examples/converters/kitti/node_modules/regenerator-runtime/runtime.js:65:40)
    at Generator.invoke [as _invoke] (/home/veith/understandai/avs-demo/xviz/examples/converters/kitti/node_modules/regenerator-runtime/runtime.js:303:22)
    at Generator.prototype.(anonymous function) [as next] (/home/veith/understandai/avs-demo/xviz/examples/converters/kitti/node_modules/regenerator-runtime/runtime.js:117:21)
    at asyncGeneratorStep (/home/veith/understandai/avs-demo/xviz/examples/converters/kitti/src/converters/base-converter.js:16:103)
    at _next (/home/veith/understandai/avs-demo/xviz/examples/converters/kitti/src/converters/base-converter.js:18:194)
    at /home/veith/understandai/avs-demo/xviz/examples/converters/kitti/src/converters/base-converter.js:18:364
    at new Promise (<anonymous>)
(node:28813) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
(node:28813) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

I spend some time trying to figure out why, and it turns out that the velodyne_points of that dataset are incomplete.
If you run 
```
cd data/kitti/2011_09_26/2011_09_26_drive_0009_sync/velodyne_points/data
seq 0 `ls | wc -l` | xargs -i sh -c "echo -n {}_;find . -name '*{}.bin' | wc -l" | grep _0 | sed -n "/_0/s/_0//p"
```

It will show you that files with the frame number
```
177
178
179
180
```
are missing. The other data streams are complete (e.g. GPS).
This is **NOT** an issue with the download script, this is a problem in the real KITTI dataset.
They are also missing if you download it by hand from the KITTI page.

This causes a mismatch in `BaseConverter.loadFrame`, because that just takes the nth index in the list of all files in the folder as that of the current frame. 
So at `frameNumber == 177`, actually the 181st velodyne_points binary file is selected, and an offset is introduced.
When it runs out of files in the end, it complains that the 443th path is undefined, even though the problem has already happened a lot earlier. 

__Possible solutions:__

- Check that the filenames are incrementally increasing in `BaseConverter.load`. This might not always be desired.
- Assert that the current frame number is contained in the selected fileName.

A better solution would also allow to add a "dummy default file", e.g. with no points (or copy the previous, or ...) for missing files.

twojtasz, 2019-03-05:
@xintongxia can you take a look at this.  Our conversion should be robust against missing data files.

pai-qi, 2021-06-16:
Is velodyne_points necessary when I want to trans kitti to xviz?