aurora-opensource/xviz #397: Create a way to reference a package directly from github

twojtasz, 2019-03-05:
We should enable users to test pre-published packages.

twojtasz, 2019-03-06:
Here are a few ideas bouncing around:
Publish build packages to a separate repo
- https://github.com/ramasilveyra/gitpkg

Publish Release (tgz) to github directly
- https://github.com/release-it/release-it

Associated with some of this is that automated changelog generation which seems a worthy addition to help automate our releases even further.

I prefer the github published artifacts with a tgz link which works with npm over another repository, but still looking into it.

jlisee, 2019-03-06:
Is the goal here to allow users instant issue resolution, for example the solution to #388 is complete, but our "monorepo" JavaScript projects cannot [feed this need](https://github.com/uber/xviz/issues/388#issuecomment-469183171):

> Not sure how I bump the version from commandline `npm install @xviz/builder` ? What version do I specify?

Because Yarn (yarnpkg/yarn#4725) and NPM (npm/npm#2974) + our build system do not allow users to install unreleased versions of our software.

So if we automate the release process it should make the process a lot faster but still not instant.  That might in fact be enough, but I was envisioning something where we can very quickly use anything landed in master.  Even having streetscape.gl itself depend on a git based XVIZ version as needed between official releases.

Does the "release-it" system solve the noise issue?  It would be awesome if we could feed bounces fixes between systems multiple times a week, but I don't think users want to wade through hundreds of releases.