aurora-opensource/xviz #399: node-expat command fails when trying to convert KITTI to XVIZ

bcsuvar, 2019-03-06:
Hi,

I get a problem trying to execute KITTI conversion to XVIZ (https://avs.auto/#/xviz/getting-started/converting-to-xviz/viewing-the-generated-xviz)

Problem seems to be with node-expat. Could someone give me a pointer in where to find the solution?

[2019-03-06T17_47_06_319Z-debug.log](https://github.com/uber/xviz/files/2937652/2019-03-06T17_47_06_319Z-debug.log)

Appreciated

twojtasz, 2019-03-06:
Looks like your on windows.

I suggest these references to see if it helps.  Ping me back either way.
https://github.com/astro/node-expat/issues/166

This is pulled in because of xml2json, we may want to move to https://www.npmjs.com/package/xml-js or https://www.npmjs.com/package/fast-xml-parser

twojtasz, 2019-03-06:
@xintongxia putting on your radar

bcsuvar, 2019-03-07:
Thanks @twojtasz, appreciate the suggestions. It hasn't worked just yet, but trying to sort it. I'm on windows, so there might be issues with that.

EliotCao, 2019-04-30:
@twojtasz Has xml2json begun to be replaced，or not?

EliotCao, 2019-05-05:
please 

[Nodejs C++ Addons](https://nodejs.org/api/addons.html)
[node-gyp installation on windows](https://github.com/nodejs/node-gyp#on-windows)
[node-gyp use](https://github.com/nodejs/node-gyp#how-to-use)

zoeeer, 2019-10-16:
I encountered the same problem.
I followed several approaches such as `npm install --global --production windows-build-tools`, but it still fails at node-gyp rebuild during `yarn`.
However, I successed by running `npm install`.