aurora-opensource/xviz #400: yarn bootstrap failed in the Quick start of XVIZ

jhx646018057, 2019-03-07:
### **when I run "yarn bootstrap" in the Quick start of XVIZ, It displayed the errors below:** 
root@ubuntu-liu:/home/AVS/xviz/scripts# yarn bootstrap
yarn run v1.12.3
$ scripts/bootstrap.sh
Bootstrapping xviz, installing in all directories

warning package-lock.json found. Your project contains lock files generated by tools other than Yarn. It is advised not to mix package managers in order to avoid resolution inconsistencies caused by unsynchronized lock files. To clear this warning, remove package-lock.json.
[1/5] Validating package.json...
[2/5] Resolving packages...
[3/5] Fetching packages...
[#----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------] 6/12483

> xviz@0.1.0 build /home/AVS/xviz
> lerna run build

lerna notice cli v3.13.1
lerna WARN project Deprecated key "commands" found in lerna.json
lerna WARN project Please rename "commands" => "command"
lerna info Executing command in 4 packages: "yarn run build"
lerna ERR! yarn run build exited 1 in '@xviz/schema'
lerna ERR! yarn run build stdout:
$ npm run clean && npm run build-schema && npm run build-protos && npm run build-es6 && npm run build-esm && npm run build-es5

> @xviz/schema@1.0.0-beta.1 clean /home/AVS/xviz/modules/schema
> rm -fr dist && mkdir -p dist

> @xviz/schema@1.0.0-beta.1 build-schema /home/AVS/xviz/modules/schema
> node ./build-schema

53 files loaded from /home/AVS/xviz/modules/schema/schema
Writing to file /home/AVS/xviz/modules/schema/dist/schema.json

> @xviz/schema@1.0.0-beta.1 build-protos /home/AVS/xviz/modules/schema
> pbjs --keep-case --target json --path ./proto/v2/*.proto --out dist/proto.v2.json

info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

lerna ERR! yarn run build stderr:
/usr/local/lib/nodejs/node-v10.15.2/out/lib/node_modules/pbjs/node_modules/protocol-buffers-schema/parse.js:87
        throw new Error('Unexpected token in message field: ' + tokens[0])
        ^

Error: Unexpected token in message field: xviz_json_schema
    at onfield (/usr/local/lib/nodejs/node-v10.15.2/out/lib/node_modules/pbjs/node_modules/protocol-buffers-schema/parse.js:87:15)
    at onmessagebody (/usr/local/lib/nodejs/node-v10.15.2/out/lib/node_modules/pbjs/node_modules/protocol-buffers-schema/parse.js:151:26)
    at onmessage (/usr/local/lib/nodejs/node-v10.15.2/out/lib/node_modules/pbjs/node_modules/protocol-buffers-schema/parse.js:200:14)
    at Function.parse (/usr/local/lib/nodejs/node-v10.15.2/out/lib/node_modules/pbjs/node_modules/protocol-buffers-schema/parse.js:573:30)
    at Object.exports.parseSchema (/usr/local/lib/nodejs/node-v10.15.2/out/lib/node_modules/pbjs/index.js:6:23)
    at Object.<anonymous> (/usr/local/lib/nodejs/node-v10.15.2/out/lib/node_modules/pbjs/cli.js:21:20)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @xviz/schema@1.0.0-beta.1 build-protos: `pbjs --keep-case --target json --path ./proto/v2/*.proto --out dist/proto.v2.json`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @xviz/schema@1.0.0-beta.1 build-protos script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-03-07T15_28_03_988Z-debug.log
error Command failed with exit code 1.

lerna ERR! yarn run build exited 1 in '@xviz/schema'
lerna WARN complete Waiting for 2 child processes to exit. CTRL-C to exit immediately.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! xviz@0.1.0 build: `lerna run build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the xviz@0.1.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-03-07T15_28_04_026Z-debug.log
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
 
**my nodejs is v10.15.2, and my ubuntu is v16.04, what should I do to resolve the error?**

twojtasz, 2019-03-07:
Need to investigate this.

rdecarreau, 2019-05-15:
Similar situation here, I run the command `yarn bootstrap` command but it says No such file or directory. Seems like there is a step missing in the quickstart. Running yarn in other directories like the examples directory just says there are no scenarios.

enginBozkurt, 2019-05-20:
First try  **sudo npm install -g yarn**
Then inside xviz folder, run "**yarn bootstrap**"

twojtasz, 2019-06-01:
@jhx646018057  you should not be running`yarn bootstrap` in the `scripts` directory.    Just in the xviz repo root.

twojtasz, 2019-06-01:
@rdecarreau Please make sure yarn is installed as @enginBozkurt explained, then run from the **root** directory. Let me know if it works or not.

DaHaiHuha, 2019-07-29:
> @rdecarreau Please make sure yarn is installed as @enginBozkurt explained, then run from the **root** directory. Let me know if it works or not.

I'm sorry but i tried and come into the same bug problem

EliotCao, 2019-08-09:
I have also a problem about 'yarn bootstrap'.
V:\Workspace\AVS\XVIZ\xviz-1.0.0-beta.19>yarn bootstrap
yarn run v1.15.2
$ yarn && ocular-bootstrap
[1/5] Validating package.json...
[2/5] Resolving packages...
[3/5] Fetching packages...
info fsevents@1.2.9: The platform "win32" is incompatible with this module.
info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from installation.
[4/5] Linking dependencies...
warning " > @babel/plugin-proposal-class-properties@7.4.4" has unmet peer dependency "@babel/core@^7.0.0-0".
info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from insta
[4/5] Linking dependencies...
warning " > @babel/plugin-proposal-class-properties@7.4.4" has unmet peer dependency "@babel/core@^7.0.
warning "@babel/plugin-proposal-class-properties > @babel/helper-create-class-features-plugin@7.4.4" ha
warning " > babel-loader@8.0.5" has unmet peer dependency "@babel/core@^7.0.0".
warning " > babel-loader@8.0.5" has unmet peer dependency "webpack@>=2".
warning " > ocular-dev-tools@0.0.27" has incorrect peer dependency "@probe.gl/test-utils@^3.0.2".
[#[######################################################################################################
######################---------------------------------------------------------------------] 21718/3386
[5/5] Building fresh packages...
warning Your current version of Yarn is out of date. The latest version is "1.17.3", while you're on "1
.15.2".
info To upgrade, download the latest installer at "https://yarnpkg.com/latest.msi".
The system cannot find the path specified.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

EliotCao, 2019-08-09:
> I have also a problem about 'yarn bootstrap'.
> V:\Workspace\AVS\XVIZ\xviz-1.0.0-beta.19>yarn bootstrap
> yarn run v1.15.2
> $ yarn && ocular-bootstrap
> [1/5] Validating package.json...
> [2/5] Resolving [...]
**> The system cannot find the path specified.
> error Command failed with exit code 1.**
> info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

it may related to schema, and I have noted that after 'yarn' or 'yarn bootstrap', there is no dist directory in module schema and other modules. but I don't know why. 
fs.js:114
    throw err;
    ^
**Error: ENOENT: no such file or directory, open 'V:\Workspace\AVS\XVIZ\xviz-1.0.0-**beta.19\modules\schema\dist\schema.json

twojtasz, 2019-08-16:
I need to get a windows machine to fix this.

enginbaglayici, 2019-08-20:
I am getting the exact same error, have you found a solution on windows machine?

zoeeer, 2019-10-10:
same error as @EliotCao  https://github.com/uber/xviz/issues/400#issuecomment-519806249 

> The system cannot find the path specified.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

OS: windows 7.
node: 10.16.0 / 12.11.1
yarn: 1.19.1

EliotCao, 2019-11-18:
yeap, I also think that it is related to windows OS, so I have change to  Linux environment to operate AVS.

rongweiSUN, 2020-05-28:
I also meet the same problem, have you found a solution on it ?

sswzfly, 2022-10-17:
I also encountered this problem,  solved by upgrading the node version from `v10.19.0` to `v16.17.1`.

ekurtgl, 2023-05-19:
> I also encountered this problem, solved by upgrading the node version from `v10.19.0` to `v16.17.1`.

@sswzfly Did "yarn bootstrap" work for Windows or did you try it on another OS?

sswzfly, 2023-05-22:
> > I also encountered this problem, solved by upgrading the node version from `v10.19.0` to `v16.17.1`.
> 
> @sswzfly Did "yarn bootstrap" work for Windows or did you try it on another OS?

I just tried it in ubuntu.

ekurtgl, 2023-05-22:
> > > I also encountered this problem, solved by upgrading the node version from `v10.19.0` to `v16.17.1`.
> > 
> > 
> > @sswzfly Did "yarn bootstrap" work for Windows or did you try it on another OS?
> 
> I just tried it in ubuntu.

I see, the problem still seems to be persistent on Windows. I tried with different node versions from 10.x to 18.x.