aurora-opensource/xviz #401: validator.validatorMetadata is not a function

tkt028, 2019-03-07:
**Context**
I follow the guideline [1] to validate the generated XVIZ data. My project is using the below package version
  -  "@xviz/builder": "^1.0.0-beta.1",
  -  "@xviz/schema": "^1.0.0-beta.1",

**Problem**
validator.validatorMetadata is not a function (please see below console log).

**Expected**
Able to validate the data according to the guideline [1]

**References**
[1] https://avs.auto/#/xviz/user-guide/validating-xviz

```
$ node
>
>
> xviz_schema = require('@xviz/schema')
{ SCHEMA_DATA: [Getter],
  XVIZValidator: [Getter],
  XVIZSessionValidator: [Getter],
  MessageTypes: [Getter],
  loadProtos: [Getter],
  getXVIZProtoTypes: [Getter],
  EXTENSION_PROPERTY: [Getter],
  protoEnumsToInts: [Getter],
  enumToIntField: [Getter],
  StructEncode: [Getter] }
> validator = new xviz_schema.XVIZValidator();
XVIZValidator {
  ajv:
   Ajv {
     _opts:
      { meta: false,
        schemaId: 'id',
        extendRefs: 'fail',
        loopRequired: Infinity,
        serialize: [Function],
        defaultMeta: 'http://json-schema.org/draft-04/schema#' },
     logger:
      Console {
        log: [Function: bound consoleCall],
        info: [Function: bound consoleCall],
        warn: [Function: bound consoleCall],
        error: [Function: bound consoleCall],
        dir: [Function: bound consoleCall],
        time: [Function: bound consoleCall],
        timeEnd: [Function: bound consoleCall],
        trace: [Function: bound consoleCall],
        assert: [Function: bound consoleCall],
        clear: [Function: bound consoleCall],
        count: [Function: bound consoleCall],
        countReset: [Function: bound countReset],
        group: [Function: bound consoleCall],
        groupCollapsed: [Function: bound consoleCall],
        groupEnd: [Function: bound consoleCall],
        Console: [Function: Console],
        debug: [Function: debug],
        dirxml: [Function: dirxml],
        table: [Function: table],
        markTimeline: [Function: markTimeline],
        profile: [Function: profile],
        profileEnd: [Function: profileEnd],
        timeline: [Function: timeline],
        timelineEnd: [Function: timelineEnd],
        timeStamp: [Function: timeStamp],
        context: [Function: context],
        [Symbol(counts)]: Map {} },
     _schemas:
[... 146 lines cut ...]
        serialize: [Function] },
     errors: null } }
>
>
> metadata = {
...   version: '2.0.0'
... };
{ version: '2.0.0' }
>
> validator.validatorMetadata(metadata);
TypeError: validator.validatorMetadata is not a function
    at repl:1:11
    at ContextifyScript.Script.runInThisContext (vm.js:50:33)
    at REPLServer.defaultEval (repl.js:240:29)
    at bound (domain.js:301:14)
    at REPLServer.runBound [as eval] (domain.js:314:12)
    at REPLServer.onLine (repl.js:441:10)
    at emitOne (events.js:121:20)
    at REPLServer.emit (events.js:211:7)
    at REPLServer.Interface._onLine (readline.js:282:10)
    at REPLServer.Interface._line (readline.js:631:8)
>
>
>
```

twojtasz, 2019-03-07:
This looks like a simple bug, could you try `validateMetadata` instead of `validatorMetadata`. I'll update the docs.

twojtasz, 2019-03-07:
https://github.com/uber/xviz/commit/aa704096f0d7592d3937fc723418e4eb8fbbadf0

Thank!

closed 2019-03-07 by twojtasz