aurora-opensource/xviz #402: Invalid lat

gerardsimons, 2019-03-07:
I get the following error trying to draw a GEOGRAPHIC coordinate type polygon. 

```
assert.js:6 Uncaught Error: invalid latitude
    at assert (assert.js:6)
    at lngLatToWorld (web-mercator-utils.js:43)
    at WebMercatorViewport.projectFlat (viewport.js:183)
    at lngLatZToWorldPosition (project-functions.js:17)
    at getWorldPosition (project-functions.js:70)
    at projectPosition (project-functions.js:120)
    at preprojectLightPositions (lighting.js:116)
    at memoize.js:35
    at ShaderModule.getUniforms [as getModuleUniforms] (lighting.js:73)
    at ShaderModule.getUniforms (shader-module.js:55)
```
This is the code I use to build the metadata:

```js
var xvizMetadataBuilder = new XVIZMetadataBuilder();

      // Exact copy of playground
      // metadata
      xvizMetadataBuilder
      .stream('/vehicle_pose')
      .category('pose')

      .stream('/ais/position')
      .category('primitive')
      .type('circle')
      .coordinate('GEOGRAPHIC')
      .streamStyle({
        fill_color: '#fb0',
        radius: 50
      })

      .stream("/vehicle/velocity")
      .category("time_series")
      .type("float")
      // TODO: How do I specify this?
      // .units("km/h")
      
      .stream("/ais/mmsi")
      .coordinate('GEOGRAPHIC')
      .category("primitive")
      .type("text")
      .streamStyle({
          "text_size": 20,
          "fill_color": "#FF0000"
      })

      .stream("/ais/trajectory")
      .category('primitive')
      .type('polyline')
      .coordinate('GEOGRAPHIC')
      .streamStyle({
        stroke_color: '#00FF00',
        stroke_width: 2,
        stroke_width_min_pixels: 1
      })

      .stream("/ais/dimension")
      .category('primitive')
      .type('polygon')
      .coordinate('GEOGRAPHIC')
      .streamStyle({
        extruded: true,
        fill_color: '#00000080'
      })
```

And this is what I use to send a frame

```
 this.xvizBuilder.pose('/vehicle_pose')
      .timestamp(Date.now())
      .mapOrigin(4.4186, 51.8889, 0)
      .position(0, 0, 0)
      .orientation(0, 0, 0);
      
      this.xvizBuilder.primitive('/ais/dimension')
      .polygon([[4.4186, 51.8889, 0], [4.5186, 51.8889, 0], [4.4186, 51.8889, 0]])
      .style({
        fill_color: '#08f',
        height: 2
      })
      .id('object-2');
```

On the streetscape side I use from getting-started > `yarn start-live` example.

Funnily enough I do not get this error when I run the exact same code in the playground (see screenshot)

[image]

twojtasz, 2019-03-07:
that assert seems to be coming from *streetscape.gl* @Pessimistress

gerardsimons, 2019-03-10:
Any quick workaround? Is the assert unnecessary? Could I remove it manually from source?

Pessimistress, 2019-03-11:
@gerardsimons A fix has been merged. If you run `yarn start-live-local` you should be able to validate it.