aurora-opensource/xviz #413: Memory leak when using polyline at high frame rate

Algirdyz, 2019-03-28:
We have built a server that uses xviz builder to convert live data into xviz format and send that to streetscape. Most is working fine, but we have a memory leak. I have been investigating it and I traced it down to one line. If I comment this line out, everything works fine and stable.
```
xvizBuilder.primitive('/ais/history')
                .polyline(hist)
                .classes(["level1"])
                .id("test");
```

Hist in this case is an array of vertices. Length maximum of 20. This is called once per trackable object. A few hundred times.
After commenting this out, we still build many other primitives for the same trackable objects, like polygons, circles, texts and so on. But no other polylines.

I have used `node-memwatch` package to investigate the heap. What I got was over 500mb of increased heap size over few minutes. Almost all of it of `Native` type (unmanaged).

The codebase is pretty large so I can't share it, but is there anything specific you need to see?

We create meta data, send it first, then we gather data form various sources over time. Every 20ms, we initiate a new xvizbuilder with the following line:
```
this.xvizBuilder = new XVIZBuilder({metaData: this.metaData});
```

then pass it along all the converters, they load the data that was gathered since the last initialization, then we call `this.xvizBuilder.getFrame();` and send it over the socket.

IF we change it, to instead create a frame every 100ms, it works find and there are no memory leaks. 

Are there limitations to xviz builder? Is it that garbage collector can't keep up or something? Would you consider this a bug?

twojtasz, 2019-03-28:
First, your setup sounds great!

Yes this sounds like a bug.  There shouldn't be any limitations outside of memory but this does seem like something is not being managed properly.  The fact that you are creating a new XVIZBuilder each time **should** means those get removed but we must be holding on to a reference somewhere most likely.

I will try to recreate this based on your description.  If you had a sample script to repo that would help a lot.  It doesn't need to be your data, just a simple case that creates the polylines from random points and exhibits the issue. Otherwise I will try to create that myself.

Also what version of Node just to make sure i'm testing as close to your setup as possible.

Algirdyz, 2019-03-29:
Thanks for looking into it. I have thought about it more and I think that this might not actually be related to polyline specifically. That one primitive takes up about half of our frame data. So most likely this is because there is too much to convert? 

For reference, each frame is about 300-500 kb.

I am using node v10.15.3. Same issue comes up on a windows 10 PC and a Macbook.

I'm sorry, but this is on a private repo. I will try to create a barebones version of this next week and upload it if still needed.

twojtasz, 2019-06-03:
There was a recent report about a bug in the websocket module we are using #466   Not sure if this is related.