aurora-opensource/xviz #419: How to transfer data between a ros application / xviz / streetscape?

baker-Xie, 2019-04-02:
I want to run a c++ ros application which will generate image/vehicle pose data and I want to display it in streetscape in realtime. I read the document but didn't find a solution. The server example only can load data in disk, but I want the server to receive data from another ros application. So what should I do ? Thanks

twojtasz, 2019-04-03:
The server example we have will only load from disk, that is correct.  But in a week or so we will have a JS websocket server that can server from effectively arbitrary data sources.  So you could, for example, have it source data from some live connection and send it through.

Now, how that  "live connection" works is kinda up to you.  We are working on doing conversion from rosbag using *rosbag.js*.  If there is a way to send message from a live system it might work, though I am not familiar enough to know the details yet.

I have heard of *roslib.js* but don't know if that connects to a live system or not.

baker-Xie, 2019-04-04:
> The server example we have will only load from disk, that is correct. But in a week or so we will have a JS websocket server that can server from effectively arbitrary data sources. So you could, [...]

Thanks. so can I using c++ to convert my data into XVIZ format, and then use Socket to send the data to StreetScape Web Client ?

twojtasz, 2019-04-29:
@baker-Xie Yes you could, but we do not have an XVIZ builder library for C++ yet.  However, you seem to get the possible data flow correctly.

pkulijing, 2019-05-05:
@baker-Xie just FYI, we are doing something similar and the following approach is working for us:
use [rosnodejs](https://www.npmjs.com/package/rosnodejs) to construct a ros node that receive ros msgs. You will also need [gennodejs](https://github.com/RethinkRobotics-opensource/gennodejs) to build your .msg files into javascript. Then you can use the nodejs sdk of xviz to transform ros msgs to xviz streams, as elaborated in the kitti example.

gennarokukonu, 2019-05-07:
@pkulijing: I am new to using AVS (XVIZ, Streetscape GL) and I am not very familiar with web based applications programming (I have primarily worked with C++ and Python in Linux).
When you mentioned in your post "Then you can use the nodejs sdk of xviz to transform ros msgs to xviz streams, as elaborated in the kitti example.", where does one find the "nodejs sdk of xviz"? I hope I am not missing the obvious. Thanks!

pkulijing, 2019-05-09:
> @pkulijing: I am new to using AVS (XVIZ, Streetscape GL) and I am not very familiar with web based applications programming (I have primarily worked with C++ and Python in Linux).
> When you mentioned [...]

By "the nodejs sdk" I mean this whole repository. For a quick start you need to understand the usage of XVIZBuilder and XVIZMetadataBuilder, which is illustrated in  [the documentation](https://avs.auto/#/xviz/user-guide/generating-xviz). Then you can refer to the [kitti example](https://github.com/uber/xviz/tree/master/examples/converters/kitti) to get a taste of how to generate real xviz frames. Pay attention to the _convertFrame_ function and _getMetadata_ function of the converters in [src/converters](https://github.com/uber/xviz/tree/master/examples/converters/kitti/src/converters). You can ignore the codes on loading and converting kitti data because they are specific to kitti. Once you use rosnodejs to obtain your ros messages in javascript format, all you need to do it to fill your data into some [geometry primitives](https://avs.auto/#/xviz/protocol/schema/geometry-primitives) with XVIZBuilder.

gennarokukonu, 2019-05-09:
@pkulijing: Thank you for your assistance!

twojtasz, 2019-07-02:
With **@xviz/ros** and the **@xviz/server** I hope we can make this better supported and documented as connecting to a live system is important.

There are some features we need to add to XVIZ/Streetscape.gl that can remove the current overhead/configuration hopefully, but that will have to come over the next few months.  I'll be working on a roadmap for this and related work and should have something in a week or two.

VijeetBenni, 2019-11-19:
How to stream the xviz frames 1 by 1 to streescape instead of  whole folder?

Frz15, 2020-04-20:
@twojtasz Thank you for your information. DId you find a way to transfer data between a ros application / xviz / streetscape? If the work is already done, could you tell me the method?