aurora-opensource/xviz #422: Support Arrays of data for time_series

twojtasz, 2019-04-03:
Issue https://github.com/uber/xviz/issues/417 wants to associate arrays & vectors of data with other streams.  This is done currently with the *id* linking in the application, but we only support scalars in [*time_series*](https://github.com/uber/xviz/blob/master/docs/protocol-schema/core-types.md#time-series-state)

We should support arrays of values.
Possibly even dictionary? not sure about this one.

twojtasz, 2019-04-03:
@jlisee will want your thoughts on this

jlisee, 2019-04-21:
@twojtasz sorry for the late reply but the we already do this through [variables](https://github.com/uber/xviz/blob/master/docs/protocol-schema/core-types.md#variables).  It would look like:

```json
{
  "base": {
    "object_id": "foo"
  },
  "values": {
    "doubles": [1001.3, 1002.3, 1003.3]
  }
}
```

I will mention this on that issue as well.