aurora-opensource/xviz #431: Can't download PDF Innvoice

dpetzer, 2019-04-26:


dpetzer, 2019-04-26:
Login to rider.uber.com
Asks me to give my location, choose skip down the bottom
Go to My Trips
Click on a trip
Click on Save Invoice 
Text changes to Loading and spins forever
[image]

Tried this in Firefox, Chrome, IE10.

Receipt is no good, need invoice.

twojtasz, 2019-04-29:
@dpetzer Sorry, this has nothing to do with https://rider.uber.com

I suggest following the instructions in the link to file a support request. https://help.uber.com/riders/article/how-to-contact-support?nodeId=4c69afef-5739-4842-9e98-ed0087b873db

closed 2019-04-29 by twojtasz

dpetzer, 2019-04-30:
@twojtasz you are a champion.