aurora-opensource/xviz #445: Cant import glb files in Blender

MatthiasThDs, 2019-05-09:
Hi

I cant import the downloaded files from streetscape.gl, neither the ones i generated myself using xviz from kitti data. Blender (www.blender.org) has an official glb importer/exporter made by the Khronos group.

Best regards
Matthias

ibgreen, 2019-05-09:
Yes, this is expected. 

- GLB is just a "binary container format" that encodes a JSON chunk and a BINARY chunk in one file. 

- Blender imports "GLB encoded" **GLTF files**.

- XVIZ uses the GLB format to efficiently encode binary data, but does not populate e.g. the GLTF `scene` object at this point.

There has been some discussion about exposing some of the XVIZ data (e.g. point clouds) as a GLTF-accessible scene graph, but it would only be a partial view.

twojtasz, 2019-05-15:
@MatthiasThDs can you elaborate on what you where hoping to do?  Knowing what you hoped to do with the GLTF would help determine what level of integration would be useful.

MatthiasThDs, 2019-05-16:
Hi,
basically I wanted to check what is inside of them. I expected to see the radar point cloud data somehow. Even though I understand now that there are multiple dataparts inside. But at that point I was not sure if everything needed had to be stored per frame. But that there is a need to store radar points for each frame was pretty clear. Anyway, a search for *.glb brought me to the Khronos website with a list of software with importers. So i gave it a try ;)
Best regards

twojtasz, 2019-05-16:
Great, thanks.

I feel pretty strong about making this standard compliant with gltf. as far as the geometry data is concerned.  One issue is that XVIZ has multiple samples over time (or could in a single message) which complicates things a bit as I know gltf supports animation but the mapping is not clear to me.  Plus XVIZ has metric data (velocity, etc) which i'm not sure how it would translate.

This is something we need to dig into more and see what maps and makes sense and what doesn't.

ibgreen, 2019-05-16:
Translating all the XVIZ geometry to glTF primitives is a lot of work it should not be hard to:
- make each point cloud into a glTF conformant mesh/scenegraph node
- we could have the timestamp as an annotation and part of the node name/id.

A good 3D viewer should show the list of scenegraph nodes in a side panel and let the user view them one by one by clicking check boxes.

- We could also create on scenegraph per timestamp, just containing the single node. Maybe worth first checking if viewers like blender can show multiple scenegraphs.

- I am not sure we should try to create a glTF animation with so much data.

twojtasz, 2019-05-20:
I agree, the animation would be overkill.

jlisee, 2019-05-23:
One change I would like to make along this lines is moving from the root `xviz` property to placing that data in the official extensions section.  This would make the JSON section of the GLB match the specification which should provide more support.

Then we can add a "compatibility mode" to the GLB writer that generates these extra scene nodes and meshes to allow for viewing the data itself in any tool.  It would be a useful feature.

This does overlap a bit with our desire @twojtasz to have a bundle format and our own tools where you can just drop a bundle (or I guess in this case a raw `state_update` message) and just see what it is in XVIZ tooling.  I feel like that is possibly the better area to place our long term investments.

ibgreen, 2019-05-23:
> moving from the root xviz property to placing that data in the official extensions section. This would make the JSON section of the GLB match the specification which should provide more support.

@jlisee @twojtasz and I went back and forth on this in the initial implementation. Our conclusion was that non-standard top-level attributes **are permitted** by glTF and we felt that the XVIZ data is not really a proper "glTF extension". But we could have gone either way.

If we are making XVIZ into an extension we could try to define what the goal is with that. We could have a global extension but that would just be an independent JSON subtree just like now, accessed by a different "key" (`extensions.xviz` instead of `xviz`).

Perhaps we could somehow define a "scene" node that has extension field that is XVIZ data so that it can be more naturally interleaved with other glTF nodes? But then we would probably want to e.g. apply the scenegraph transformation matrix.

twojtasz, 2019-05-24:
@ib while permitted, other libraries may not support arbitrary top-level entries.  So since it has extensions and extras they seem appropriate and a minor structural change.

The **extra** vs **extension** i thought we discussed before but lost the thread. I recall leaning toward extra as this is "application" specific data.  However, if we have plans (which we seem to) to integrate more in the glTF scene and nodes in some fashion, then an extension seems appropriate.

However, I don't know the node structure well enough to know if it makes sense to put it there or just as the blob in an `extensions.xviz` location

ibgreen, 2019-05-24:
> However, I don't know the node structure well enough to know if it makes sense to put it there or just as the blob in an extensions.xviz location

@twojtasz Top-level extension should be fine for now. Node-level can be added later and should be more granular.

twojtasz, 2022-01-02:
gltf will not be supported in 1.x.

closed 2022-01-02 by twojtasz