aurora-opensource/xviz #450: version v1.0.0-beta.9 results in error in lidar points generation

pkulijing, 2019-05-14:
I was using version v1.0.0-beta.8 and my lidar data could be generated correctly as a `point` primitive, similar to what is done in the kitti converter example. But after upgrading to v1.0.0-beta.9, the lidar points are absent, and the related part in the generated glb file is null:
[image]
In v1.0.0-beta.8 that same part was something like this:
[image]

I noticed that version v1.0.0-beta.9 included a commit related to flatten arrays, in which the lidar-converter.js file in the kitti converter example was modified. Is this a bug or a backward-incompatible change? If the latter is the case, then I think the [documentation ](https://avs.auto/#/xviz/getting-started/converting-to-xviz/mapping-data-to-xviz/lidar?section=lidar-scans)needs to be updated.

twojtasz, 2019-05-15:
@pkulijing I believe this is a bug that is fixed in https://github.com/uber/xviz/commit/8252902afbaf14a13b706a379f1660f084217d6b

I am waiting for another change to land then I can release a new beta.

twojtasz, 2019-05-15:
I just published **v1.0.0-beta.13**. Please update and try again. Thanks!

twojtasz, 2019-06-01:
@pkulijing I believe this is fixed and was a problem with just the beta.9 release

closed 2019-06-01 by twojtasz

pkulijing, 2019-06-13:
I can confirm that the problem no longer exists in beta.16.