aurora-opensource/xviz #452: Xviz example server crash

truewhite, 2019-05-15:
When I run run-kitti-example.sh to open an xviz server and then try to run the yarn start-streaming under streetscape.gl/examples/get-started, the xviz server crashes with the following error: Why is this?

```c++
Loading frames from /home/chen/cpp/self/uber/xviz/scripts/../data/generated/kitti/2011_09_26/2011_09_26_drive_0005_sync/
Checking for files:  /home/chen/cpp/self/uber/xviz/data/generated/kitti/2011_09_26/2011_09_26_drive_0005_sync/1-frame.glb
Checking for index file
Loaded 154 frames
[id:1] > Connection from Client.
/home/chen/cpp/self/uber/xviz/node_modules/@loaders.gl/gltf/dist/es5/gltf/gltf-parser.js:140
      var data = this.json[key];
                          ^

TypeError: Cannot read property 'xviz' of undefined
```

twojtasz, 2019-05-15:
It should not. Seems a problem with the data it is expecting.  In this case it is trying to decode the saved GLB files, which should have a JSON chunk. Inside that object it should have an 'xviz' key, but here `this.json` is undefined for some reason.

Could you upload the 1-frame.glb and 2-frame.glb from your `generated/kitti/....` directory?

We are working on tools that will allow you to quickly dump data and test for yourself.

twojtasz, 2019-05-15:
You may want to try with **v1.0.0-beta.13**. It fixed a few other bugs and with luck may help you.

truewhite, 2019-05-16:
Here is my 1-frame.glb and 2-frame.glb file，I tried v1.0.0-beta.13 but it failed.
[2011_09_26_drive_0005_sync.tar.gz](https://github.com/uber/xviz/files/3185206/2011_09_26_drive_0005_sync.tar.gz)

twojtasz, 2019-05-16:
@truewhite ok, so I believe this is related to `@loaders.gl` which was recently upgraded and a problem it has with `TextDecoder`. 

I ran into this and used node v11 and the problem went away (also I did a `yarn bootstrap` to make sure everything was up-to-date).

Can you try with node 11 and let me know. In the meantime i'll see if I can get `@loaders.gl` fixed

truewhite, 2019-05-17:
When I use node v11, the xviz server no longer crashes. I used node v8 before.

But I suspect this data conversion is problematic because the visualization of the streetscape.gl client is not what I expected.

Below is the display of the streetscape.gl client
[image]

twojtasz, 2019-05-17:
@truewhite thanks for validating.  I have not seen this math.gl error.  I feel the streetscape.gl issue is likely something else (which I need to try to repo).

Also, this is highlighting a need to be able to validate XVIZ and serving w/o streetscape.gl, which I will be working on in the next phase (starting in a few weeks).

The only suggestion I can make would be to do the same for streetscape.gl.  fresh sync and run `yarn bootstrap`.

Actually if you can give me the specific xviz & streetscape.gl git ref's then I will try to repo

truewhite, 2019-05-20:
ok, thanks for your assistance, maybe there is a problem somewhere, I will try it again.

xuluxi, 2019-05-29:
I am encounter the same problem

YunjieYu, 2019-05-30:
Hello, 
I am also experiencing the same issue.  My environment is Ubuntu16.04, node 10.
When I follow @truewhite's operations, i.e., running ./scripts/run-kitti-example.sh to convert KITTI data and open an xviz server, and then running the yarn start-streaming under streetscape.gl/examples/get-started, the xviz server crashes with the same error: TypeError: Cannot read property 'xviz' of undefined.

Furthermore, I follow the tutorial in https://avs.auto/#/xviz/getting-started/converting-to-xviz/viewing-the-generated-xviz and https://avs.auto/#/streetscape.gl/getting-started/starter-kit:
1. Convert KITTI to XVIZ Format
$ cd examples/converters/kitti
$ yarn
$ yarn start -d 2011_09_26/2011_09_26_drive_0051_sync -o 2011_09_26/2011_09_26_drive_0051_sync --json --image-max-width=600  // if I disenable --json, this will generate glb files and then cause the same error: TypeError: Cannot read property 'xviz' of undefined.
2. Load the XVIZ extracted from the KITTI dataset
2.1 I copied the generated "2011_09_26_drive_0051_sync" files (0-frame.json and xxx-frame.json) to streetscape.gl/examples/get-started/.
2.2 I changed the streetscape.gl/examples/get-started/src/log-from-file.js to:

export default new XVIZFileLoader({
  timingsFilePath:
  './2011_09_26_drive_0051_sync/0-frame.json',
  // 'https://raw.githubusercontent.com/uber/xviz-data/master/kitti/2011_09_26_drive_0005_sync/0-frame.json',
  getFilePath: index =>
  `./2011_09_26_drive_0051_sync/${index + 1}-frame.json`,
  // `https://raw.githubusercontent.com/uber/xviz-data/master/kitti/2011_09_26_drive_0005_sync/${index + 1}-frame.glb`,
  worker: true,
  maxConcurrency: 4
});

2.3 yarn start // start the app

This leads to the same case again. streetscape.gl can only display the vehicle data, and there is no point cloud data.

Any quick workaround?

YunjieYu, 2019-05-30:
Hello,
I am also experiencing the same issue. My environment is Ubuntu16.04, node 10.
When I follow @truewhite's operations, i.e., running ./scripts/run-kitti-example.sh to convert KITTI data and open an xviz server, and then running the yarn start-streaming under streetscape.gl/examples/get-started, the xviz server crashes with the same error: TypeError: Cannot read property 'xviz' of undefined.

Furthermore, I follow the tutorial in https://avs.auto/#/xviz/getting-started/converting-to-xviz/viewing-the-generated-xviz and https://avs.auto/#/streetscape.gl/getting-started/starter-kit:

1. Convert KITTI to XVIZ Format
$ cd examples/converters/kitti
$ yarn
$ yarn start -d 2011_09_26/2011_09_26_drive_0051_sync -o 2011_09_26/2011_09_26_drive_0051_sync --json --image-max-width=600 // if I disenable --json, this will generate glb files and then cause the same error: TypeError: Cannot read property 'xviz' of undefined.
2. Load the XVIZ extracted from the KITTI dataset
2.1 I copied the generated "2011_09_26_drive_0051_sync" files (0-frame.json and xxx-frame.json) to streetscape.gl/examples/get-started/.
2.2 I changed the streetscape.gl/examples/get-started/src/log-from-file.js to:
export default new XVIZFileLoader({
timingsFilePath:
'./2011_09_26_drive_0051_sync/0-frame.json',
// 'https://raw.githubusercontent.com/uber/xviz-data/master/kitti/2011_09_26_drive_0005_sync/0-frame.json',
getFilePath: index =>
./2011_09_26_drive_0051_sync/${index + 1}-frame.json,
// https://raw.githubusercontent.com/uber/xviz-data/master/kitti/2011_09_26_drive_0005_sync/${index + 1}-frame.glb,
worker: true,
maxConcurrency: 4
});

2.3 yarn start // start the app

This leads to the same case again. streetscape.gl can only display the vehicle data, and there is no point cloud data.

Any quick workaround?

YunjieYu, 2019-05-30:
Hello,
I am also experiencing the same issue. My environment is Ubuntu16.04, node 10.
When I follow @truewhite's operations, i.e., running ./scripts/run-kitti-example.sh to convert KITTI data and open an xviz server, and then running the yarn start-streaming under streetscape.gl/examples/get-started, the xviz server crashes with the same error: TypeError: Cannot read property 'xviz' of undefined.

Furthermore, I follow the tutorial in https://avs.auto/#/xviz/getting-started/converting-to-xviz/viewing-the-generated-xviz and https://avs.auto/#/streetscape.gl/getting-started/starter-kit:

Convert KITTI to XVIZ Format
$ cd examples/converters/kitti
$ yarn
$ yarn start -d 2011_09_26/2011_09_26_drive_0051_sync -o 2011_09_26/2011_09_26_drive_0051_sync --json --image-max-width=600 // if I disenable --json, this will generate glb files and then cause the same error: TypeError: Cannot read property 'xviz' of undefined.
Load the XVIZ extracted from the KITTI dataset
2.1 I copied the generated "2011_09_26_drive_0051_sync" files (0-frame.json and xxx-frame.json) to streetscape.gl/examples/get-started/.
2.2 I changed the streetscape.gl/examples/get-started/src/log-from-file.js to:
export default new XVIZFileLoader({
timingsFilePath:
'./2011_09_26_drive_0051_sync/0-frame.json',
// 'https://raw.githubusercontent.com/uber/xviz-data/master/kitti/2011_09_26_drive_0005_sync/0-frame.json',
getFilePath: index =>
./2011_09_26_drive_0051_sync/${index + 1}-frame.json,
// https://raw.githubusercontent.com/uber/xviz-data/master/kitti/2011_09_26_drive_0005_sync/${index + 1}-frame.glb,
worker: true,
maxConcurrency: 4
});

    2.3 yarn start // start the app

This leads to the same case again. streetscape.gl can only display the vehicle data, and there is no point cloud data.

Any quick workaround?

YunjieYu, 2019-05-30:
Hello,
I am also experiencing the same issue. My environment is Ubuntu16.04, node 10.
When I follow @truewhite's operations, i.e., running ./scripts/run-kitti-example.sh to convert KITTI data and open an xviz server, and then running the yarn start-streaming under streetscape.gl/examples/get-started, the xviz server crashes with the same error: TypeError: Cannot read property 'xviz' of undefined.

Furthermore, I follow the tutorial in https://avs.auto/#/xviz/getting-started/converting-to-xviz/viewing-the-generated-xviz and https://avs.auto/#/streetscape.gl/getting-started/starter-kit:

1. Convert KITTI to XVIZ Format
$ cd examples/converters/kitti
$ yarn
$ yarn start -d 2011_09_26/2011_09_26_drive_0051_sync -o 2011_09_26/2011_09_26_drive_0051_sync --json --image-max-width=600 
// Note that if I disenable --json, this will generate glb files and then cause the same error: TypeError: Cannot read property 'xviz' of undefined.
2. Load the XVIZ extracted from the KITTI dataset
2.1 I copied the generated "2011_09_26_drive_0051_sync" files (0-frame.json and xxx-frame.json) to streetscape.gl/examples/get-started/.
2.2 I changed the streetscape.gl/examples/get-started/src/log-from-file.js to:
export default new XVIZFileLoader({
timingsFilePath:
'./2011_09_26_drive_0051_sync/0-frame.json',
// 'https://raw.githubusercontent.com/uber/xviz-data/master/kitti/2011_09_26_drive_0005_sync/0-frame.json',
getFilePath: index =>
./2011_09_26_drive_0051_sync/${index + 1}-frame.json,
// https://raw.githubusercontent.com/uber/xviz-data/master/kitti/2011_09_26_drive_0005_sync/${index + 1}-frame.glb,
worker: true,
maxConcurrency: 4
});

    2.3 yarn start // start the app

This leads to the same case again. streetscape.gl can only display the vehicle data, and there is no point cloud data.

Any quick workaround?

YunjieYu, 2019-05-30:
## Result
[image]

upadsha, 2020-01-03:
> You may want to try with **v1.0.0-beta.13**. It fixed a few other bugs and with luck may help you.

Server crashes on sending JSON object messages to xvizBuilder with .image() format mismatch, even on converting it to Uint8Array, on any UDP packets mismatch delivery, missing header with height & width, makes it crashes. 
Any solution for this ?