aurora-opensource/xviz #454: Change from **Frame** to **Message**

twojtasz, 2019-05-16:
The concept of **Frame** was originally a "frame" of data (or a data frame).
Normally we are talking about a **message**, (general XVIZ message which can be metadata or state_update, in a more expanded view could be any XVIZ message like start, transform_log etc)

Therefore to align with the rest of the XVIZ spec lets move from **Frame** -> **Message**.

Here is a list of places where we reference **Frame**
[ ] builder **getFrame**()
[ ] xvizWriter **writeFrame**()
[ ] synchronizer **getCurrentFrame**()
[ ] XVIZ general docs refer to "frame"

# Where does **Frame** still make sense?
It may make sense to leave the **converters** since those API's are not part of the public API and we are conceptually making a "movie" out of data.  A movie is made up of "Frames" so if you squint it makes some sense semantically where Frame seems better than Message.

Pessimistress, 2019-05-16:
I think it's ok for the synchronizer to keep using `frame` because what's returned is an aggregated result from multiple messages. Either way we need to differentiate this concept from `message`.

twojtasz, 2019-05-16:
@Pessimistress I agree.  I'll update on exactly what I plan to change to Message and what will remain Frame.

jlisee, 2019-05-23:
Yeah I like the split on `frame` is a full picture of the XVIZ world while a `message` is just an update we are getting or creating. In the spec we have a `state_update` which contains `stream_set` a message is actually can either be a `state_update` or a `metadata` or several other things.  So one possible option is:

[ ] builder **getMessage**()
[ ] xvizWriter **writeMessage**()
[ ] synchronizer **getCurrentFrame**()
[ ] XVIZ general docs refer to messages explicitly

twojtasz, 2019-05-24:
@jlisee i'm plan on going with what you proposed, and providing deprecated functions to avoid breakage but they will not be documented.

twojtasz, 2019-05-28:
https://github.com/uber/xviz/pull/467 implements this change

closed 2019-05-28 by twojtasz