aurora-opensource/xviz #455: Error when using converters

xuzidi, 2019-05-16:
~/project/xviz/examples/converters/kitti$ yarn start -d ~/project/xviz/data/027/ -o ~/project/xviz/data/07
yarn run v1.15.2
warning package.json: No license field
$ node ./src/index.js -d /home/autowise/project/xviz/data/027/ -o /home/autowise/project/xviz/data/07
/home/autowise/project/xviz/data/027/ /home/autowise/project/xviz/data/07
Converting KITTI data at /home/autowise/project/xviz/data/027/
Saving to /home/autowise/project/xviz/data/07
processing gps data frame 0/188
*** Error in `/usr/local/bin/node': corrupted size vs. prev_size: 0x00007f32e40011f0 ***
*** Error in `/usr/local/bin/node': corrupted size vs. prev_size: 0x00007f32dc0011f0 ***
*** Error in `/usr/local/bin/node': corrupted size vs. prev_size: 0x00007f32d40011f0 ***

twojtasz, 2019-05-16:
this msg appears to be a glibc msg.  What version of node are you using?  You may want to try +/- 1 major version.

Can you make a minimal test case?  that way I can reproduce and look into.  It is unclear if this is KITTI or if you might just be reusing that code for your own data.

xuzidi, 2019-05-17:
> this msg appears to be a glibc msg. What version of node are you using? You may want to try +/- 1 major version.
> 
> Can you make a minimal test case? that way I can reproduce and look into. It is unclear if this is KITTI or if you might just be reusing that code for your own data.

Thanks for replying.
1. I used KITTI data set.
2. That error happened using node -v 10. * Then I updated node to the latest version v12.2.0 and ran yarn and a new error happened. 
info fsevents@1.2.9: The platform "linux" is incompatible with this module.
info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
[4/4] Building fresh packages...
[1/2] ⠂ sharp
error /home/autowise/project/xviz/examples/converters/kitti/node_modules/sharp: Command failed.
Exit code: 1
Command: (node install/libvips && node install/dll-copy && prebuild-install) || (node-gyp rebuild && node install/dll-copy)
Arguments: 
Directory: /home/autowise/project/xviz/examples/converters/kitti/node_modules/sharp
Output:
info sharp Detected globally-installed libvips v8.7.4
info sharp Building from source via node-gyp

Package libgsf-1 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libgsf-1.pc'
to the PKG_CONFIG_PATH environment variable
Package 'libgsf-1', required by 'vips', not found
gyp: Call to 'PKG_CONFIG_PATH="/opt/ros/kinetic/lib/pkgconfig:/opt/ros/kinetic/lib/x86_64-linux-gnu/pkgconfig:/usr/local/lib/pkgconfig:/usr/lib/pkgconfig" pkg-config --cflags-only-I vips-cpp vips glib-2.0 | sed s\/-I//g' returned exit status 0 while in binding.gyp. while trying to load binding.gyp

3.Another question: Can I use rosbag data to do the visualization? If I use the rosbag data, do I need to
write new converters ?  
Thanks very much!

twojtasz, 2019-05-17:
@xuzidi  Thanks for that info, I need to do a round of linux testing and will try to follow this flow.

For rosbag we will be releasing a module (next week) to assist in visualizing ROS.  Ideally you will get some value out of the default setup, but there will be some configuration and customization for messages we don't handle by default.

truewhite, 2019-05-20:
@twojtasz ok, Thank you for your assistance, I think it may be a problem with my data source, let me try it again.