aurora-opensource/xviz #457: Set the number of points in the point cloud

tower111, 2019-05-20:
All point cloud loading speed is too slow, I need to load some points, how do I need to set it?
Is there a parameter available?
thank you

twojtasz, 2019-05-20:
@tower111 What data set are you using?  Is this KITTI or your own?  You would have to sample the data.  There is no option to do this currently in any existing code but the conversion code is relatively short and easy to modify.

tower111, 2019-05-21:
Using kitti data (downloaded from the official website), where is the conversion code?

tower111, 2019-05-21:
> @ tower111您使用的数据集是什么？这是KITTI还是你自己的？您必须对数据进行采样。目前在任何现有代码中都没有选项，但转换代码相对较短且易于修改。

tower111, 2019-05-21:
@twojtasz Using kitti data (downloaded from the official website), where is the conversion code?

twojtasz, 2019-06-01:
@tower111  https://avs.auto/#/xviz/getting-started/converting-to-xviz/mapping-data-to-xviz/lidar describe the lidar data conversion.

You can find the conversion code @ https://github.com/uber/xviz/tree/master/examples/converters/kitti

tower111, 2019-06-09:
> @tower111 https://avs.auto/#/xviz/getting-started/converting-to-xviz/mapping-data-to-xviz/lidar describe the lidar data conversion.
> 
> You can find the conversion code @ https://github.com/uber/xviz/tree/master/examples/converters/kitti
thanks.
I found it below
Xviz-master/examples/converters/kitti/src/parsers/parse-lidar-points.js

I changed the contents of the file to the following.
STEP is the number of points to display.
I think the better way to improve it is to have fewer points in the distance and more in the near.

```
// Copyright (c) 2019 Uber Technologies, Inc.
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * Parse LiDar data (stored in velodyne_points dir),
 */

import {Parser as BinaryParser} from 'binary-parser';
const parser = new BinaryParser().floatle();
 var STEP=5*4;
function readBinaryData(binary) {
  const res = [];

  for (let i = 0; i < binary.length; i = i+4 ) {
    if (i + 4 > binary.length) {
      break;
    }
    const parsed = parser.parse(binary.slice(i, i +4));
    res.push(parsed);
  }
  return res;
}
export function loadLidarData(data) {
 
  const binary = readBinaryData(data);
  const float = new Float32Array(binary);
  const size = Math.round(binary.length / STEP);

  // We could return interleaved buffers, no conversion!
  const positions = new Float32Array(3 * size);
  const colors = new Uint8Array(4 *  size).fill(255);

//dian de mi du
  for (let i = 0; i < size; i++) {
    positions[i * 3 + 0] = float[i * STEP + 0];
    positions[i * 3 + 1] = float[i * STEP + 1];
    positions[i * 3 + 2] = float[i * STEP + 2];

    const reflectance = Math.min(float[i * 4 + 3], 3);//zhe li
    colors[i * 4 + 0] = 80 + reflectance * 80;
    colors[i * 4 + 1] = 80 + reflectance * 80;
    colors[i * 4 + 2] = 80 + reflectance * 60;
    //you xie dian hui bu xian shi
  }
  return {positions, colors};
}
```