aurora-opensource/xviz #465: Examples to load pointcloud data

speaktobrijesh, 2019-05-27:
Hi,

Currently working with Aeye and we have the requirement to load the point cloud data in XVIZ and to visualize to streetscape.gl application.
currently we have the point cloud data in float32Array. Looking for any examples how to load the point cloud data in XVIZ as its currently coming from live sensor.  

Thanks.

pkulijing, 2019-05-28:
Here is the code I used to transform lidar data contained in ros msg in `/sensor_msgs/PointCloud2` format ([doc](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/PointCloud2.html))

The key is to select from your data the xyz and rbga of the points. If your data only contains an intensity value, you should make up an rgba like I do below.

    load_lidar_data(lidar_msg) {
        let points_binary = [];
        let intensity_binary = [];
        for (let i = 0; i < lidar_msg.data.length; i += lidar_msg.point_step) {
          const x = parser.parse(lidar_msg.data.slice(i, i + 4));
          const y = parser.parse(lidar_msg.data.slice(i + 4, i + 8));
          const z = parser.parse(lidar_msg.data.slice(i + 8, i + 12));
          points_binary.push(x);
          points_binary.push(y);
          points_binary.push(z);
          const intensity = parser.parse(lidar_msg.data.slice(i + 16, i + 20));
          intensity_binary.push(intensity);
        }
        const points = new Float32Array(points_binary);
        let colors = [];
        const intensity_float = new Float32Array(intensity_binary);    
        for (let i = 0; i < intensity_float.length; ++i) {
            colors.push(255 - intensity_float[i]); // r
            colors.push(intensity_float[i]); // g
            colors.push(0); // b
            colors.push(255); // a
        }
        return {points, colors};
    }

    get_frame(builder, lidar_msg, frame_cnt) {
        if (!lidar_msg) {
        console.log("lidar-msg: no lidar_msg");
        return;
        }
        const t1 = Date.now();
        const lidar_data = this.load_lidar_data(lidar_msg);
        const t2 = Date.now();
        console.log(`frame %d: lidar data loading took %d ms`,
        frame_cnt, 
        t2 - t1);
        builder
        .primitive(this.LIDAR_POINTS)
        .points(lidar_data.points)
        .colors(lidar_data.colors)
    }

twojtasz, 2019-06-01:
@speaktobrijesh This may help https://avs.auto/#/xviz/getting-started/converting-to-xviz/mapping-data-to-xviz/lidar

Are you looking for something ROS related or more general?