aurora-opensource/xviz #466: Serious memory leak caused by websocket package

pkulijing, 2019-05-28:
The example server uses [ws](https://www.npmjs.com/package/ws) to implement the websocket server. I wrote my own live server based on the example, and noticed that the server suffers from serious memory leak when network bandwidth is insufficient. 
The issue is not difficult to reproduce. Run the server in live mode, and let multiple clients (i.e. streetscape.gl front end) connect to it. When the bandwidth required totally by all the clients exceeds the maximum supported by your device (the network interface, the router, etc), the memory occupied by the server will keep increasing, eventually crashes the machine. 

At the beginning, the process occupied 5% of my total memory:
[image]

After openning enough clients to exhaust the network bandwidth, the memory footprint began to boost:
[image]
[image]
[image]
Even if I closed all the clients now, the memory footprint never decreased:
[image]

After some googling, I found that the  [ws](https://www.npmjs.com/package/ws) package has [a historical memory leak problem](https://github.com/websockets/ws/issues?utf8=%E2%9C%93&q=memory+leak) , which has not been fully resolved today. I know it's unbelievable: a widely-used package with 10000+ github stars has an easy-to-reproduce memory leak.

I switched to the [uWS](https://github.com/uNetworking/uWebSockets)package and it worked like a charm. The memory footprint stays below 8% when the bandwidth is exhausted:
[image]

And drops back to 5% after I close all the clients:
[image]
 
uWS is only my personal choice, but I think ws should definitely be avoided due to the memory leak. 

Here is my uWS server implementation in case it may help. The only thing worthy of note is the `idleTimeout` option. uWS requires that the server receive some msg from the client every `idleTimeout` seconds. If no msg arrives, the client is considered dead and closed by the server. Since streetscape.gl frontend does not send any message to the server, I set the number to a extremely large number to prevent the client from being closed. 

    const uWS = require('uWebSockets.js/uws.js');
    const app = uWS.App()
      .ws('/*', {
        /* Options */
        compression: 0,
        maxPayloadLength: 16 * 1024 * 1024,
        idleTimeout: 100000000,
        /* Handlers */
        open: (ws, req) => {
          console.log('A WebSocket connected via URL: ' + req.getUrl());
          this.ws_clients.add(ws);
          this.send_data_one_client(ws, this.metadata_frame);
          console.log("sent metadata");
        },
        message: (ws, message, isBinary) => {
          console.log("server receives some msg");
        },
        drain: (ws) => {
          console.log('drain: WebSocket backpressure: ' + ws.getBufferedAmount());
        },
        close: (ws, code, message) => {
          console.log(`WebSocket closed: code ${code}, message: ${message}`);
          this.ws_clients.delete(ws);
        }
      }).any('/*', (res, req) => {
        res.end('Nothing to see here!');
      }).listen(port, (token) => {
        if (token) {
          console.log('Listening to port ' + port);
        } else {
          console.log('Failed to listen to port 10000' + port);
        }
      });
    send_data_one_client(ws, data) {
      ws.send(data, true);
    }

    send_data_all_clients(data) {
      this.ws_clients.forEach(client => {
        this.send_data_one_client(client, data);
      });
    }

twojtasz, 2019-05-28:
Thanks for the details.  I was planning on allowing the server implemention to be swapped out (by providing an instance of another server.  Looks like I should accelerate this feature and look at other websocket server implementations.

alexhultman, 2019-09-05:
If you set the idleTimeout to 0 it is disabled. However you really should use it - behavior can be changed to allow the server to send pings (not as distributed as the server need to to more work, but hey).

Other than that I agree, ws is lacking. Feel free to contact me for quotation / brainstorming.