aurora-opensource/xviz #469: KIITI to XVIZ fails

jrabary, 2019-05-30:
I'm trying to convert kitti by following the documentation but it fails with the following error
```
internal/modules/cjs/loader.js:583
    throw err;
    ^

Error: Cannot find module 'babel-polyfill'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/[...]/xviz/examples/converters/kitti/src/index.js:16:1)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

pkulijing, 2019-05-30:
I think you need to run `npm install` in `example/converters/kitti`

twojtasz, 2019-06-01:
@jrabary What this the **Quick Start** instructions?  I realized the script to run the kitti setup didn't run `yarn` so I added it.  That should avoid this happening in to others. Thanks!

closed 2019-06-01 by twojtasz

jrabary, 2019-06-02:
@twojtasz Thanks, indeed running yarn was missing in the documentation.

EvanLind, 2019-07-10:
@twojtasz I run `yarn` or `yarn install`,and it returns *done*,but there still cannot' find module './dist/es5/node'
```shell
kitti git:(master) ✗ yarn install
yarn install v1.16.0
warning package.json: No license field
warning jskitti@1.1.0: No license field
[1/4] Resolving packages...
success Already up-to-date.
Done in 0.25s.

➜  kitti git:(master) ✗ yarn start -d ../../../../xviz-data/kitti/2011_09_26_drive_0005_sync
yarn run v1.16.0
warning package.json: No license field
$ node ./src/index.js -d ../../../../xviz-data/kitti/2011_09_26_drive_0005_sync
internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module './dist/es5/node'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:690:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/home/evanlin/Documents/src/AVS/xviz/modules/io/node.js:14:18)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Module._compile (/home/evanlin/Documents/src/AVS/xviz/examples/converters/kitti/node_modules/pirates/lib/index.js:99:24)
    at Module._extensions..js (internal/modules/cjs/loader.js:787:10)
    at Object.newLoader [as .js] (/home/evanlin/Documents/src/AVS/xviz/examples/converters/kitti/node_modules/pirates/lib/index.js:104:7)
    at Module.load (internal/modules/cjs/loader.js:653:32)
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

EliotCao, 2019-08-05:
> @twojtasz I run `yarn` or `yarn install`,and it returns _done_,but there still cannot' find module './dist/es5/node'. Thanks
> 
> ```shell
> kitti git:(master) ✗ yarn install
> yarn install v1.16.0
> warning package.json: No license field
> warning jskitti@1.1.0: No license field
> [1/4] Resolving packages...
> success Already up-to-date.
> Done in 0.25s.
> 
> ➜  kitti git:(master) ✗ yarn start -d ../../../../xviz-data/kitti/2011_09_26_drive_0005_sync
> yarn run v1.16.0
> warning package.json: No license field
> $ node ./src/index.js -d ../../../../xviz-data/kitti/2011_09_26_drive_0005_sync
> internal/modules/cjs/loader.js:638
>     throw err;
>     ^
> 
> Error: Cannot find module './dist/es5/node'
>     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
>     at Function.Module._load (internal/modules/cjs/loader.js:562:25)
>     at Module.require (internal/modules/cjs/loader.js:690:17)
>     at require (internal/modules/cjs/helpers.js:25:18)
>     at Object.<anonymous> (/home/evanlin/Documents/src/AVS/xviz/modules/io/node.js:14:18)
>     at Module._compile (internal/modules/cjs/loader.js:776:30)
>     at Module._compile (/home/evanlin/Documents/src/AVS/xviz/examples/converters/kitti/node_modules/pirates/lib/index.js:99:24)
>     at Module._extensions..js (internal/modules/cjs/loader.js:787:10)
>     at Object.newLoader [as .js] (/home/evanlin/Documents/src/AVS/xviz/examples/converters/kitti/node_modules/pirates/lib/index.js:104:7)
>     at Module.load (internal/modules/cjs/loader.js:653:32)
> error Command failed with exit code 1.
> info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
> ```

have you solved this problem?

EliotCao, 2019-08-05:
> @twojtasz I run `yarn` or `yarn install`,and it returns _done_,but there still cannot' find module './dist/es5/node'. Thanks
>
> ```shell
> kitti git:(master) ✗ yarn install
> yarn install v1.16.0
> warning [...]

This problem is caused by missing XVIZ/IO ,you could deal it by add @xviz/builder to the KITTI converter example's package.json. like as 'yarn add -D @xviz/builder' in kitti example foloder.

EliotCao, 2019-08-05:
> @twojtasz I run `yarn` or `yarn install`,and it returns _done_,but there still cannot' find module './dist/es5/node'. Thanks
>
> ```shell
> kitti git:(master) ✗ yarn install
> yarn install v1.16.0
> warning [...]

The problem is caused by missing XVIZ/IO, you could deal it by add @xviz/builder to the KITTI converter example's package.json like as "yarn add -D @xviz/builder".

bldulam1, 2020-03-25:
"yarn add -D @xviz/builder"

solves the problem