aurora-opensource/xviz #470: What is the recommended version of nodejs?

pkulijing, 2019-05-30:
I've been using node 11 in my development. Since it's not a LTS, I decide to downgrade to node 10. But the app no longer works.  The following code provides a minimal reproduce of the problem:

    'use strict';
    require('@babel/register');
    require('babel-polyfill');

    const XVIZWriter = require('@xviz/builder').XVIZWriter;
    const XVIZMetadataBuilder = require('@xviz/builder').XVIZMetadataBuilder;

    const writer = new XVIZWriter();
    const builder = new XVIZMetadataBuilder();
    builder
      .stream("/vehicle_pose")
      .category('pose')

    const metadata = builder.getMetadata();
    writer.writeMetadata("/tmp", metadata);

With node 11 it works fine and generates `/tmp/1-frame.glb` successfully. But with node 10 it results in the following error:

    /home/caros/Developer/test/node_modules/@loaders.gl/gltf/dist/es5/glb/glb-builder.js:204
        var textEncoder = new _core.TextEncoder('utf8');
                            ^

    TypeError: _core.TextEncoder is not a constructor
        at GLTFBuilder._convertObjectToJsonChunk (/home/caros/Developer/test/node_modules/@loaders.gl/gltf/dist/es5/glb/glb-builder.js:204:25)
        at GLTFBuilder._createGLBBuffer (/home/caros/Developer/test/node_modules/@loaders.gl/gltf/dist/es5/glb/glb-builder.js:161:28)
        at GLTFBuilder.encodeAsGLB (/home/caros/Developer/test/node_modules/@loaders.gl/gltf/dist/es5/glb/glb-builder.js:67:19)
        at encodeBinaryXVIZ (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-binary-writer.js:34:22)
        at writeBinaryXVIZtoFile (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-binary-writer.js:38:23)
        at XVIZWriter.writeMetadata (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-writer.js:104:53)
        at Object.<anonymous> (/home/caros/Developer/test/main.js:15:8)
        at Module._compile (internal/modules/cjs/loader.js:776:30)
        at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
        at Module.load (internal/modules/cjs/loader.js:653:32)

The project is attached.
[test.tar.gz](https://github.com/uber/xviz/files/3237396/test.tar.gz)

However, the kitti example still works fine. Maybe I've done something wrong in my project, but I   really fail to see it. Could you please provide some help?

twojtasz, 2019-05-30:
I will try your example. I would expect that our modules include the appropriate polyfill where necessary, although you may be able to work around it by including it in your project for the time being.

ibgreen, 2019-05-30:
> I would expect that our modules include the appropriate polyfill where necessary,

Summary: Yes, going forward, my recommendation is that XVIZ modules should import `@loaders.gl/polyfills` directly for any modules that are mainly intended to run under node.

Long story:

The initial version of `2loaders.gl/core` automatically installed the `TextEncoder` polyfills under Node.js, however the polyfills we use are quite big and there were subtle bundler-dependent issues that sometimes caused unnecessary polyfills to be bundled in the browser which caused hard to solve issues, so we moved them to a separate polyfills module so that the user is in control. 

There is information on this in the v1.1. notes in https://loaders.gl/docs/upgrade-guide, however it is belated since these changes happened as hot-fixes in the v1.0.x release series.

In retrospect we should probably have done this fix as a major version bump rather than a patch.

pkulijing, 2019-05-31:
> I will try your example. I would expect that our modules include the appropriate polyfill where necessary, although you may be able to work around it by including it in your project for the time being.

Thanks for the help. I required `@loaders.gl/polyfills` in my project and now it works.

twojtasz, 2019-05-31:
I was able to reproduce. I'm waiting on some dependent changes and should have a fix in for our next beta release. in a day or so

twojtasz, 2019-06-01:
@pkulijing I tested with the latest with node v10 and I didn't run into the problem.  We updated a dependency that I believe may have been causing an issue with some globals.

closed 2019-06-01 by twojtasz