aurora-opensource/xviz #478: Live streaming source converter example or specification needed

AndyBoat, 2019-06-05:
Hi I am an soft engineer interested in this great project. 
I've noticed  the word "Real-Time  Transfer and Visualization of Autonomy Data"  in ur website and  ur spec also named out that a running system source  is supported , but I cant find any example or spec about how to create a converter from a running system source in this project, while there are two types of  static log file converter in this project . 
Could u please give me some guide about how to build a converter for a running system source, because I  want to apply this amazing tool on a running vehicle system.
Highest respect.

twojtasz, 2019-06-20:
Since this system requires a *conversion* to XVIZ, there has to be a piece of software that performs this conversion somewhere.  That is *Step 1*.  *Step 2* is the transferring of that data to a client, in our examples we use a websocket server to communicate with.

With those satisfied you can stream live data to the web application. If your platform is ROS based, various people on our slack room have mentioned they have made this work with https://github.com/RobotWebTools/roslibjs, though I have not had the chance to setup a system to replicate this.

We are working on some MVP ROS Support, but it will not cover the live conversion case yet just the BAG support.  However it will provide a structure and server that should make that live support a bit easier.

cmpute, 2019-07-01:
Hi @twojtasz , thanks for you work on this. I want to write a python server to serve our data and I meet similar problem. Is there any documentation for live data stream? Does the data feed from websocket differ from ones contained in the `.glb` files? Actually as long as the protocol is clear for live stream, I can quickly build a python server/converter.