aurora-opensource/xviz #480: RangeError: Offset is outside the bounds of the DataView

pkulijing, 2019-06-13:
I came accross this error after upgrading to the newest version from 1.0.0-beta.8. I tried to google the problem and it seems to be a long-existing problem in the nodejs community. I will try to construct a minimal reproduction ASAP.

    RangeError: Offset is outside the bounds of the DataView
        at DataView.getUint32 (<anonymous>)
        at getBmpSize (/home/julienjli/Developer/xviz-converter/node_modules/@loaders.gl/images/dist/es5/lib/image-parsers.js:53:22)
        at getImageSize (/home/julienjli/Developer/xviz-converter/node_modules/@loaders.gl/images/dist/es5/lib/get-image-metadata.js:73:14)
        at getImageMetadata (/home/julienjli/Developer/xviz-converter/node_modules/@loaders.gl/images/dist/es5/lib/get-image-metadata.js:85:18)
        at GLTFBuilder.addImage (/home/julienjli/Developer/xviz-converter/node_modules/@loaders.gl/gltf/dist/es5/lib/deprecated/gltf-builder.js:204:54)
        at packBinaryJsonTypedArray (/home/julienjli/Developer/xviz-converter/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:16:34)
        at packBinaryJson (/home/julienjli/Developer/xviz-converter/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:55:12)
        at packBinaryJson (/home/julienjli/Developer/xviz-converter/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:62:24)
        at /home/julienjli/Developer/xviz-converter/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:49:16
        at Array.map (<anonymous>)

pkulijing, 2019-06-14:
[test.tar.gz](https://github.com/uber/xviz/files/3289945/test.tar.gz)

I'm able to reproduce the issue with the following piece of code:

    import {XVIZBuilder, XVIZMetadataBuilder} from '@xviz/builder';
    import {XVIZBinaryWriter} from '@xviz/io';
    import {FileSink} from '@xviz/io/node';

    const builder = new XVIZBuilder();
    const xvizWriter = new XVIZBinaryWriter(new FileSink("/tmp"));

    const line1 = [
      [ -41.21190834, 10.265042, 0 ],
      [ -43.239524, 10.415601, 0 ]
    ];

    const line2 = [
      [ -41.211908341, 10.265042, 0 ],
      [ -43.239524, 10.415601, 0 ]
    ];

    builder
      .pose('/vehicle_pose')
      .timestamp(1)
      .mapOrigin(0, 0, 0);

    builder
      .primitive("/apolyline")
      .polyline(line1);

    xvizWriter.writeMessage(1, builder.getMessage());
    console.log("line1 written successfully");

    builder
      .primitive("/apolyline")
      .polyline(line2);

    xvizWriter.writeMessage(2, builder.getMessage());
    console.log("line2 written successfully");

The output is:
    caros@ubuntu:~/Developer/test$ node -r esm test.js 
    TextEncoder constructor called with encoding label, which is ignored.
    TextEncoder constructor called with encoding label, which is ignored.
    line1 written successfully
    Stream /apolyline primitives vertices are not provided.
    /home/caros/Developer/test/node_modules/@loaders.gl/images/dist/es5/lib/get-image-metadata.js:1
    Error: invalid image data for type: image/jpeg
        at getImageSize (/home/caros/Developer/test/node_modules/@loaders.gl/images/dist/es5/lib/get-image-metadata.js:76:11)
        at getImageMetadata (/home/caros/Developer/test/node_modules/@loaders.gl/images/dist/es5/lib/get-image-metadata.js:85:18)
        at GLTFBuilder.addImage (/home/caros/Developer/test/node_modules/@loaders.gl/gltf/dist/es5/lib/deprecated/gltf-builder.js:204:54)
        at packBinaryJsonTypedArray (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:16:34)
        at packBinaryJson (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:55:12)
        at packBinaryJson (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:62:24)
        at /home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:49:16
        at Array.map (<anonymous>)
        at packBinaryJson (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:48:21)
        at packBinaryJson (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:62:24)

It seems that adding one more digit to a coordinate of a vertex makes a big difference.

The minimum reproduction project is attached. I'm using `esm`  so the script should be run with `node -r esm test.js`.

pkulijing, 2019-06-14:
What's more, the `TextEncoder constructor called with encoding label, which is ignored.` warning message is kind of annoying.

twojtasz, 2019-06-20:
@pkulijing yes, I know.  I"ll try to remove that message shortly.  It is in a dependency of ours now

twojtasz, 2019-06-20:
This is weird, thanks for the repro case.

roarcurry, 2019-07-09:
I also have met this problem. 

When I run the kitti converter example: `yarn start -d <path> -o <path>`, it comes error:
```
(node:1531) UnhandledPromiseRejectionWarning: Error: Input file is missing or of an unsupported image format
(node:1531) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:1531) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Done in 1.61s.
```

then I disabled image streams like `yarn start -d <path> -o <path> --disable-streams=image_00,image_01,image_02,image_03`. it started well at first, when progress comes to 90/150,  then error occured:
```
(node:1416) UnhandledPromiseRejectionWarning: Error: invalid image data for type: image/jpeg
    at getImageSize (/home/cidi/Desktop/xviz/examples/converters/kitti/node_modules/@loaders.gl/images/src/lib/get-image-metadata.js:49:11)
    at getImageSize (/home/cidi/Desktop/xviz/examples/converters/kitti/node_modules/@loaders.gl/images/src/lib/get-image-metadata.js:60:20)
    at GLTFBuilder.addImage (/home/cidi/Desktop/xviz/examples/converters/kitti/node_modules/@loaders.gl/gltf/src/lib/deprecated/gltf-builder.js:206:25)
    at addImage (/home/cidi/Desktop/xviz/examples/converters/kitti/node_modules/@xviz/builder/src/writers/xviz-writer/xviz-pack-binary.js:19:36)
    at packBinaryJsonTypedArray (/home/cidi/Desktop/xviz/examples/converters/kitti/node_modules/@xviz/builder/src/writers/xviz-writer/xviz-pack-binary.js:54:12)
    at packBinaryJson (/home/cidi/Desktop/xviz/examples/converters/kitti/node_modules/@xviz/builder/src/writers/xviz-writer/xviz-pack-binary.js:60:24)
    at packBinaryJson (/home/cidi/Desktop/xviz/examples/converters/kitti/node_modules/@xviz/builder/src/writers/xviz-writer/xviz-pack-binary.js:48:36)
    at Array.map (<anonymous>)
    at map (/home/cidi/Desktop/xviz/examples/converters/kitti/node_modules/@xviz/builder/src/writers/xviz-writer/xviz-pack-binary.js:48:21)
    at packBinaryJson (/home/cidi/Desktop/xviz/examples/converters/kitti/node_modules/@xviz/builder/src/writers/xviz-writer/xviz-pack-binary.js:60:24)
(node:1416) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:1416) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Done in 41.30s.
```

I use absolute path and it is correct. 
Hope for reply, Thanks!

Algirdyz, 2019-08-12:
I'm having this issue. Any tips on workaround until fixed? I've tried rounding my coordinate numbers but it just reduces the frequency of the error. I get it eventually anyway.

pkulijing, 2019-08-12:
> I'm having this issue. Any tips on workaround until fixed? I've tried rounding my coordinate numbers but it just reduces the frequency of the error. I get it eventually anyway.

I tried but was unable to walk around it. I'm sticking to version v1.0.0-beta.8 before a solution is provided.

twojtasz, 2019-08-12:
So the original issue is caused because the encoding of the float value -41.211910247802734 as 32-bit float is 0xC224D8FF in hex.  When stored in little-endian that is store as 0xFF 0xD8 0x24 0xC2.  When we read the buffer to *detect* if the data is an image we get a false positive and thus the reset of the errors.  (huge 🙇 to @pkulijing for the repro case)

I have a fix for @loaders.gl/image that will make the jpeg detection a little bit more robust.  Another round of changes will have to add more intelligence into the XVIZ binary writer to be more intelligent about the types of streams and data in the streams it is converting so it does not create such false positives.

Will update once the fix is published (1 more day I think)

twojtasz, 2019-08-12:
Verified the fix locally, but there are other number patterns that can trigger a false positive.  I need to look in to adding more smarts.  I want to see if this address the current active issues and then prioritize the general fix based on that.

twojtasz, 2019-08-15:
@pkulijing @Algirdyz @roarcurry this is fixed in **1.0.0-beta.20**
Please try it out and let me know if you are still experiencing problems.

pkulijing, 2019-08-28:
@twojtasz Using the 1.0.0 version, I'm not seeing the original exception. But there still exists something wrong with drawing polylines. I'm seeing a new exception frequently:

    RangeError: Offset is outside the bounds of the DataView
        at DataView.getUint32 (<anonymous>)
        at getBmpSize (/home/jing/Developer/xviz-converter/node_modules/@loaders.gl/images/dist/es5/lib/image-parsers.js:53:22)
        at getImageSize (/home/jing/Developer/xviz-converter/node_modules/@loaders.gl/images/dist/es5/lib/get-image-metadata.js:73:14)
        at getImageMetadata (/home/jing/Developer/xviz-converter/node_modules/@loaders.gl/images/dist/es5/lib/get-image-metadata.js:85:18)
        at GLTFBuilder.addImage (/home/jing/Developer/xviz-converter/node_modules/@loaders.gl/gltf/dist/es5/lib/deprecated/gltf-builder.js:204:54)
        at packBinaryJsonTypedArray (/home/jing/Developer/xviz-converter/node_modules/@xviz/io/dist/es5/writers/xviz-pack-binary.js:16:34)
        at packBinaryJson (/home/jing/Developer/xviz-converter/node_modules/@xviz/io/dist/es5/writers/xviz-pack-binary.js:55:12)
        at packBinaryJson (/home/jing/Developer/xviz-converter/node_modules/@xviz/io/dist/es5/writers/xviz-pack-binary.js:62:24)
        at /home/jing/Developer/xviz-converter/node_modules/@xviz/io/dist/es5/writers/xviz-pack-binary.js:49:16
        at Array.map (<anonymous>)

pkulijing, 2019-08-28:
In case it may help, I'm able to find a minimal reproduction of the problem:
[test.tar.gz](https://github.com/uber/xviz/files/3551102/test.tar.gz)

In short, it happens when I try to draw a polyline with vertices 

[[ 27.28772277585972, 11.732284622432305, 0 ],
[ 25.364206288204883, 11.777934187517136, 0 ]]

Algirdyz, 2019-10-23:
I am also getting this without BMP.
```RangeError: Offset is outside the bounds of the DataView
    at DataView.getUint32 (<anonymous>)
    at getUint32 (/home/algis/Documents/xviz-visualizer/node_modules/@xviz/builder/node_modules/@loaders.gl/images/src/lib/image-parsers.js:54:21)
    at getSize (/home/algis/Documents/xviz-visualizer/node_modules/@xviz/builder/node_modules/@loaders.gl/images/src/lib/get-image-metadata.js:46:16)
    at getImageSize (/home/algis/Documents/xviz-visualizer/node_modules/@xviz/builder/node_modules/@loaders.gl/images/src/lib/get-image-metadata.js:60:20)
    at GLTFBuilder.addImage (/home/algis/Documents/xviz-visualizer/node_modules/@xviz/builder/node_modules/@loaders.gl/gltf/src/lib/deprecated/gltf-builder.js:206:25)```
I am not using any bmp or images in general in this build.

ibgreen, 2019-10-23:
@Algirdyz Are you able to see what the content of (or perhaps more importantly, length of) your data buffer is when this error happens? 

Is your data buffer very short? I wonder if it crashes because the BMP file detection doesn't check the length first?

upadsha, 2020-01-03:
> I'm able to reproduce the issue with the following piece of code:
> 
> ```
> import {XVIZBuilder, XVIZMetadataBuilder} from '@xviz/builder';
> import {XVIZBinaryWriter} from '@xviz/io';
> import {FileSink} from '@xviz/io/node';
> 
> const builder = new XVIZBuilder();
> const xvizWriter = new XVIZBinaryWriter(new FileSink("/tmp"));
> 
> 
> const line1 = [
>   [ -41.21190834, 10.265042, 0 ],
>   [ -43.239524, 10.415601, 0 ]
> ];
> 
> const line2 = [
>   [ -41.211908341, 10.265042, 0 ],
>   [ -43.239524, 10.415601, 0 ]
> ];
> 
> 
> builder
>   .pose('/vehicle_pose')
>   .timestamp(1)
>   .mapOrigin(0, 0, 0);
> 
> 
> builder
>   .primitive("/apolyline")
>   .polyline(line1);
> 
> xvizWriter.writeMessage(1, builder.getMessage());
> console.log("line1 written successfully");
> 
> builder
>   .primitive("/apolyline")
>   .polyline(line2);
> 
> xvizWriter.writeMessage(2, builder.getMessage());
> console.log("line2 written successfully");
> ```
> 
> The output is:
> 
> ```
> caros@ubuntu:~/Developer/test$ node -r esm test.js 
> TextEncoder constructor called with encoding label, which is ignored.
> TextEncoder constructor called with encoding label, which is ignored.
> line1 written successfully
> Stream /apolyline primitives vertices are not provided.
> /home/caros/Developer/test/node_modules/@loaders.gl/images/dist/es5/lib/get-image-metadata.js:1
> Error: invalid image data for type: image/jpeg
>     at getImageSize (/home/caros/Developer/test/node_modules/@loaders.gl/images/dist/es5/lib/get-image-metadata.js:76:11)
>     at getImageMetadata (/home/caros/Developer/test/node_modules/@loaders.gl/images/dist/es5/lib/get-image-metadata.js:85:18)
>     at GLTFBuilder.addImage (/home/caros/Developer/test/node_modules/@loaders.gl/gltf/dist/es5/lib/deprecated/gltf-builder.js:204:54)
>     at packBinaryJsonTypedArray (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:16:34)
>     at packBinaryJson (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:55:12)
>     at packBinaryJson (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:62:24)
>     at /home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:49:16
>     at Array.map (<anonymous>)
>     at packBinaryJson (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:48:21)
>     at packBinaryJson (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:62:24)
> ```
> 
> It seems that adding one more digit to a coordinate of a vertex makes a big difference.
> 
> The minimum reproduction project is attached. I'm using `esm` so the script should be run with `node -r esm test.js`.
> 
> [test.tar.gz](https://github.com/uber/xviz/files/3289945/test.tar.gz)

Hi i am getting similar issue / error message, for converting JSON object to xviz BINARY_GLB format or even Uint8Array buffer data.

upadsha, 2020-01-03:
> I also have met this problem.
>
> When I run the kitti converter example: `yarn start -d <path> -o <path>`, it comes error:
>
> ```
> (node:1531) UnhandledPromiseRejectionWarning: Error: Input file is [...]

converting the frame to glb using the xviz writer

upadsha, 2020-01-03:
> I came across this error after upgrading to the newest version from 1.0.0-beta.8.
> 
> ```
> Error: invalid image data for type: image/jpeg
>     at getImageSize (/home/caros/Developer/test/node_modules/@loaders.gl/images/dist/es5/lib/get-image-metadata.js:76:11)
>     at getImageMetadata (/home/caros/Developer/test/node_modules/@loaders.gl/images/dist/es5/lib/get-image-metadata.js:85:18)
>     at GLTFBuilder.addImage (/home/caros/Developer/test/node_modules/@loaders.gl/gltf/dist/es5/lib/deprecated/gltf-builder.js:204:54)
>     at packBinaryJsonTypedArray (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:16:34)
>     at packBinaryJson (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:55:12)
>     at packBinaryJson (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:62:24)
>     at /home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:49:16
>     at Array.map (<anonymous>)
>     at packBinaryJson (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:48:21)
>     at packBinaryJson (/home/caros/Developer/test/node_modules/@xviz/builder/dist/es5/writers/xviz-writer/xviz-pack-binary.js:62:24)
> ```

converting the frame to glb using the xviz writer