aurora-opensource/xviz #483: How to efficiently deliver real-time image data？

truewhite, 2019-06-21:
I tried to pass the real-time camera image to the xviz server, but the bmp image is too big. Is there any good way to pass the data quickly?

truewhite, 2019-06-21:
The way I can think of is to perform jpeg compression on the image, although the encoding requires a certain amount of time cost, affecting the real-time nature of the system.

twojtasz, 2019-06-21:
@truewhite This is always going to be a challenging problem.  The best options are to encode efficiently, and reduce size (size or resolution).  Of course, this assumes you are sending the data in the binary efficient GLB format and not JSON with base64 encoding.

twojtasz, 2019-07-08:
@truewhite Is there anything else you think to be done on this issue?   I will restate that sending as png/jpeg with the GLB format should greatly improve your performance.  However, now matter what you do, there will be some lag introduced, so that really needs quantified and then measured for your use-case.   Is 100ms delay ok, for example.

jlisee, 2019-07-12:
I also recommend a two level approach:
 - By default send a reduced resolution and/or lower rate image to provide context
 - When needed send a configuration command to your server to have it switch into high resolution mode, generally for just a single image/camera stream instead of all of them

In reality most users are looking at images at a reduced resolution on screen, so it does not make sense to send the raw large image over.

truewhite, 2019-07-13:
What I am doing now is to reduce the resolution to transfer images. I didn't think of other good methods. The general jpg/png encoding is too slow. If you can speed up encoding through gpu, it might be better, but it has not been done yet.

truewhite, 2019-07-13:
@jlisee This may be a good approach, and I am currently using a way to reduce the resolution.

jlisee, 2019-07-14:
> The general jpg/png encoding is too slow.

We have not tried this but since any image the browser supports should work, you should be able to send `bmp` images too.  Which don't require any image compression, just at most pixel format conversion, so it should be considerably faster, at the cost of a 2-10x bandwidth jump.