aurora-opensource/xviz #489: Is there any specifications about Map rendering?

cmpute, 2019-07-03:
Currently XVIZ protocol only offer ability to transfer vehicle GPS location, and streetscape.gl render the base map on Mapbox. However map from Mapbox is not precise enough for AV, is there any way to render our own high-definition map (e.g. with clear lane info)? If not, is there any plan to support this?

I'm not sure whether this issue is related to XVIZ or streetscape.gl. I think it would be better to support custom map rendering from XVIZ protocol side.

twojtasz, 2019-07-08:
We do have an example for how to extend the streetscape.gl app to add a new layer that can display additional data, such as GeoJSON.

https://avs.auto/#/streetscape.gl/advanced-topics/add-custom-3d-layers?section=render-custom-map-layers

There is a feature to add persistent data which one use-case is for vector based rendering of map features using the existing XVIZ primitives.  This can go far, but is ultimately limiting if you need more complex state management of the map data.  Specifically, the purging of map data when it is no longer contextually necessary (too far away) is important.  The state management of data at this level is a bit outside of XVIZ for the time being, though we are looking into the best way to ensure HD mapping can be supported.  It may not be XVIZ, but it is something that needs to work easily in the streetscape.gl application.

I would be curious to know specifics like format of the data, area coverage and if purging (for space or performance reasons) is required.  Map annotations are also important but those can be done with simple geospatial markers independent of the map data itself.

cmpute, 2019-07-08:
Thanks for your reply. Actually by now I didn't need to render a large map in that scale, currently I just want to render a small map for simulators (I use CARLA). In CARLA, the map is in OpenDrive format. Besides, other platforms, such as Autoware, have their own vector HD map.

I didn't notice that advance topic, thanks for referring that. However, that may be a bit hard for me since I'm not quite familiar with Javascript. Is there a way to easily change a map render (such as Google Map or Open Street Map)?

twojtasz, 2019-07-16:
We are hoping to find a way to support open drive.

The easiest would be to get the opendrive into geojson and then add that to the app per the instructions above.  However there is still some complexity to working out details as I have not done it myself yet.

If you give it a go ping me with any issues and I'll try to help.

Longer term (~2months) I hope to have some info on open drive or transforming the code for CARLA

pai-qi, 2021-04-20:
> Thanks for your reply. Actually by now I didn't need to render a large map in that scale, currently I just want to render a small map for simulators (I use CARLA). In CARLA, the map is in OpenDrive [...]

Hi, May I ask what progress you have made yet? I`m having the same problem about xodr now...

AyaYaf, 2021-06-07:
> Thanks for your reply. Actually by now I didn't need to render a large map in that scale, currently I just want to render a small map for simulators (I use CARLA). In CARLA, the map is in OpenDrive [...]

Hi, May I ask what progress you have made yet? I`m having the same problem about OSM now...

AyaYaf, 2021-06-07:
> Hi, May I ask what progress you have made yet? I`m having the same problem about xodr now...

Hi, May I ask what progress you have made yet? I`m having the same problem about OSM now...

pai-qi, 2021-06-18:
Hi,I'm sorry I didn't reply because the project schedule.
I'm still at the stage that integrates the HD map and pointcloud map，it`s the future goal to change the map render.

JianXinyu, 2021-12-13:
> We are hoping to find a way to support open drive.
>
> The easiest would be to get the opendrive into geojson and then add that to the app per the instructions above. However there is still some [...]

Could you please give some advice on how to convert the XODR format into GeoJson? Thanks!

octan123, 2021-12-13:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。