aurora-opensource/xviz #490: Why can the client receive only one .glb file when live mode is turned on?

zyqqqq, 2019-07-05:
When both server and client are turned on in live mode, the client can only receive one .glb file(metadata), But the remaining .glb file(state_update)  cannot be receive . How can I receive all. GLB files?  And Why still need a JSON file to record timestamps when Live mode is turned on？

Here are the commands I used to turn on live mode
[image]
[image]

twojtasz, 2019-07-08:
That should not be the case. Let me try to reproduce

roarcurry, 2019-07-11:
It seems that the instance of XVIZLiveLoader in `streetscape.gl/examples/get-started/src/log-from-live.js`  lacks of one property "session_type: 'live'".
Because in line 120 of `xviz/modules/server/src/server/xviz-provider-session.js`,  `params.session_type` is `undefined`.

zyqqqq, 2019-07-12:
I did it!!!!  thank you very much

closed 2019-07-12 by zyqqqq