aurora-opensource/xviz #492: Run ROS example

LGlayama, 2019-07-08:
I try to follow the Basic ROS Example to convert ros data to rviz. But I failed to see the visualization at  http://localhost:8080/kitti_2011_09_26_drive_0005_synced. The error message show "Cannot GET /kitti_2011_09_26_drive_0005_synce".

I converted kitti data to ros bag, generate the config file and then replace it as the example 1, is there anything wrong?

twojtasz, 2019-07-16:
@LGlayama can you check the url.

You should see an attempt on the server terminal to connections and if it could not find something.
On the Browser if you open the developer console you may see something in the console window, or in the Network -> WS panels (in Chrome) with more information about the websocket connection adn messages.

Any screenshot on this would help.

twojtasz, 2019-08-16:
Hi @LGlayama I also noticed  the path should be  '/kitti_2011_09_26_drive_0005_synced', but your error says **Cannot GET /kitti_2011_09_26_drive_0005_synce**

Please reopen this if you still are having issues.

closed 2019-08-16 by twojtasz