aurora-opensource/xviz #501: Error when using xvizros configuration

jasoues, 2019-07-21:
~/xviz/examples/converters/ros$ node index.js -d ../../../data/kitti/ --rosConfig config-1.json
internal/modules/cjs/loader.js:628
    throw err;
    ^

Error: Cannot find module '../build/Release/sharp.node'
Require stack:
- /home/jasser/xviz/node_modules/sharp/lib/constructor.js
- /home/jasser/xviz/node_modules/sharp/lib/index.js
- /home/jasser/xviz/modules/ros/dist/es5/messages/sensor-image-converter.js
- /home/jasser/xviz/modules/ros/dist/es5/messages/index.js
- /home/jasser/xviz/modules/ros/dist/es5/providers/register.js
- /home/jasser/xviz/modules/ros/dist/es5/index.js
- /home/jasser/xviz/examples/converters/ros/main.js
- /home/jasser/xviz/examples/converters/ros/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:625:15)
    at Function.Module._load (internal/modules/cjs/loader.js:527:27)
    at Module.require (internal/modules/cjs/loader.js:683:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/jasser/xviz/node_modules/sharp/lib/constructor.js:10:15)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Module._compile (/home/jasser/xviz/examples/converters/ros/node_modules/pirates/lib/index.js:99:24)
    at Module._extensions..js (internal/modules/cjs/loader.js:787:10)
    at Object.newLoader [as .js] (/home/jasser/xviz/examples/converters/ros/node_modules/pirates/lib/index.js:104:7)
    at Module.load (internal/modules/cjs/loader.js:643:32) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/jasser/xviz/node_modules/sharp/lib/constructor.js',
    '/home/jasser/xviz/node_modules/sharp/lib/index.js',
    '/home/jasser/xviz/modules/ros/dist/es5/messages/sensor-image-converter.js',
    '/home/jasser/xviz/modules/ros/dist/es5/messages/index.js',
    '/home/jasser/xviz/modules/ros/dist/es5/providers/register.js',
    '/home/jasser/xviz/modules/ros/dist/es5/index.js',
    '/home/jasser/xviz/examples/converters/ros/main.js',
    '/home/jasser/xviz/examples/converters/ros/index.js'
  ]
}

jasoues, 2019-07-28:
Uninstalling node.js 12 and installing node.js 10 resolved the problem

twojtasz, 2019-08-16:
Thanks.  Sharp does seem sensitive to the node version.  It seems node 10.15 works the the current sharp, but if you go to 10.16, we need sharp 0.23.

I'll be adding this to the FAQ until it is fixed.

closed 2019-08-16 by twojtasz