aurora-opensource/xviz #506: yarn bootstrap failed

DaHaiHuha, 2019-07-29:
As in the issue #400, I've come to the similar problem
First I install Node.js and npm and yarn (all the latest versions), the versions are as follows 
```shell
$ npm -v && node -v && yarn --version
6.10.0
v12.7.0
1.17.3
```
then I run the command `yarn bootstrap` and I got below:
```shell
$ yarn cache clean
yarn cache v1.17.3
success Cleared cache.
Done in 2.90s.

# alfredo @ alfredo-Ubuntu in ~/gits/xviz on git:master x [16:04:42] 
$ yarn run clean  
yarn run v1.17.3
error Command "clean" not found.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

# alfredo @ alfredo-Ubuntu in ~/gits/xviz on git:master x [16:05:44] C:1
$ npm run clean
npm ERR! missing script: clean

# alfredo @ alfredo-Ubuntu in ~/gits/xviz on git:master x [16:05:51] C:1
$ yarn bootstrap
yarn run v1.17.3
$ yarn && ocular-bootstrap
warning package-lock.json found. Your project contains lock files generated by tools other than Yarn. It is advised not to mix package managers in order to avoid resolution inconsistencies caused by unsynchronized lock files. To clear this warning, remove package-lock.json.
[1/5] Validating package.json...
[2/5] Resolving packages...
[3/5] Fetching packages...
info fsevents@1.2.9: The platform "linux" is incompatible with this module.
info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from installation.
[4/5] Linking dependencies...
warning " > bootstrap@4.3.1" has unmet peer dependency "jquery@1.9.1 - 3".
warning " > bootstrap@4.3.1" has unmet peer dependency "popper.js@^1.14.7".
warning " > @babel/plugin-proposal-class-properties@7.4.4" has unmet peer dependency "@babel/core@^7.0.0-0".
warning "@babel/plugin-proposal-class-properties > @babel/helper-create-class-features-plugin@7.4.4" has unmet peer dependency "@babel/core@^7.0.0".
warning " > babel-loader@8.0.5" has unmet peer dependency "@babel/core@^7.0.0".
warning " > babel-loader@8.0.5" has unmet peer dependency "webpack@>=2".
[5/5] Building fresh packages...
[7/7] ⡀ sharp
[2/7] ⡀ puppeteer
[-/7] ⡀ waiting...
[-/7] ⡀ waiting...
error /home/alfredo/gits/xviz/node_modules/sharp: Command failed.
Exit code: 1
Command: (node install/libvips && node install/dll-copy && prebuild-install) || (node-gyp rebuild && node install/dll-copy)
Arguments: 
Directory: /home/alfredo/gits/xviz/node_modules/sharp
Output:
info sharp Using cached /home/alfredo/.npm/_libvips/libvips-8.6.1-linux-x64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=12.7.0 runtime=node arch=x64 platform=linux)
gyp info it worked if it ends with ok
gyp info using node-gyp@4.0.0
gyp info using node@12.7.0 | linux | x64
gyp info spawn /usr/bin/python2
gyp info spawn args [
gyp info spawn args   '/home/alfredo/gits/xviz/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/home/alfredo/gits/xviz/node_modules/sharp/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/alfredo/gits/xviz/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/alfredo/.node-gyp/12.7.0/include/node/common.gypi',
[... 173 lines cut ...]
     v8::Handle<v8::Object> input, std::vector<v8::Local<v8::Object>> &buffersToPersist
                                                                      ^
../src/common.cc:43:71: error: ‘buffersToPersist’ was not declared in this scope
     v8::Handle<v8::Object> input, std::vector<v8::Local<v8::Object>> &buffersToPersist
                                                                       ^
../src/common.cc:609:1: error: expected ‘}’ at end of input
 }  // namespace sharp
 ^
sharp.target.mk:132: recipe for target 'Release/obj.target/sharp/src/common.o' failed
make: Leaving directory '/home/alfredo/gits/xviz/node_modules/sharp/build'
make: *** [Release/obj.target/sharp/src/common.o] Error 1
gyp ERR! build error 
gyp ERR! stack **Error: `make` failed with exit code: 2**
gyp ERR! stack     at ChildProcess.onExit (/home/alfredo/gits/xviz/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:203:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Linux 4.15.0-54-generic
gyp ERR! command "/opt/node-v12.7.0-linux-x64/bin/node" "/home/alfredo/gits/xviz/node_modules/.bin/node-gyp" "rebuild"
gyp ERR! cwd /home/alfredo/gits/xviz/node_modules/sharp
gyp ERR! node -v v12.7.0

```
This is on the Ubuntu16.04 and sincerely hope and thanks that anyone who can give me a solution and i'm willing to contrubute to this project as much as I can, this tool is cool

@twojtasz this is another problem from previous one, could you please give me any suggestion for start use? I'm new at JavaScript and Web technics but interested in this project

dks333, 2019-07-29:
Same issues. Any advice?

dks333, 2019-07-29:
Try it on node 10.16.0

DaHaiHuha, 2019-07-30:
> Try it on node 10.16.0

no good

# alfredo @ alfredo-Ubuntu in ~/gits/xviz on git:master x [9:38:41]
$ node -v && nodejs -v && npm -v && yarn -v
v10.16.0
v10.16.0
6.9.0
1.17.3

# alfredo @ alfredo-Ubuntu in ~/gits/xviz on git:master x [9:38:43]
$ yarn bootstrap                           
yarn run v1.17.3
$ yarn && ocular-bootstrap
warning package-lock.json found. Your project contains lock files generated by tools other than Yarn. It is advised not to mix package managers in order to avoid resolution inconsistencies caused by unsynchronized lock files. To clear this warning, remove package-lock.json.
[1/5] Validating package.json...
[2/5] Resolving packages...
[3/5] Fetching packages...
info fsevents@1.2.9: The platform "linux" is incompatible with this module.
info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from installation.
[4/5] Linking dependencies...
warning " > bootstrap@4.3.1" has unmet peer dependency "jquery@1.9.1 - 3".
warning " > bootstrap@4.3.1" has unmet peer dependency "popper.js@^1.14.7".
warning " > @babel/plugin-proposal-class-properties@7.4.4" has unmet peer dependency "@babel/core@^7.0.0-0".
warning "@babel/plugin-proposal-class-properties > @babel/helper-create-class-features-plugin@7.4.4" has unmet peer dependency "@babel/core@^7.0.0".
warning " > babel-loader@8.0.5" has unmet peer dependency "@babel/core@^7.0.0".
warning " > babel-loader@8.0.5" has unmet peer dependency "webpack@>=2".
[5/5] Building fresh packages...
[6/7] ⠐ sharp
[2/7] ⠐ puppeteer
[-/7] ⠐ waiting...
[-/7] ⠐ waiting...
error /home/alfredo/gits/xviz/node_modules/sharp: Command failed.
Exit code: 1
Command: (node install/libvips && node install/dll-copy && prebuild-install) || (node-gyp rebuild && node install/dll-copy)
Arguments:
Directory: /home/alfredo/gits/xviz/node_modules/sharp
Output:
info sharp Using cached /home/alfredo/.npm/_libvips/libvips-8.6.1-linux-x64.tar.gz
(node:7927) UnhandledPromiseRejectionWarning: ZlibError: zlib: unexpected end of file
    at Unzip.Zlib.(anonymous function) (/home/alfredo/gits/xviz/node_modules/minizlib/index.js:126:21)
    at Unzip.write (/home/alfredo/gits/xviz/node_modules/minizlib/index.js:247:21)
    at Unzip.flush (/home/alfredo/gits/xviz/node_modules/minizlib/index.js:206:10)
    at Unzip.end (/home/alfredo/gits/xviz/node_modules/minizlib/index.js:213:10)
    at Unpack.end (/home/alfredo/gits/xviz/node_modules/sharp/node_modules/tar/lib/parse.js:416:21)
    at Yallist.pipes.forEach.p (/home/alfredo/gits/xviz/node_modules/minipass/index.js:288:18)
    at Yallist.forEach (/home/alfredo/gits/xviz/node_modules/yallist/yallist.js:154:8)
    at ReadStream.emit (/home/alfredo/gits/xviz/node_modules/minipass/index.js:285:18)
    at ReadStream.emit (/home/alfredo/gits/xviz/node_modules/fs-minipass/index.js:153:22)
    at ReadStream.[maybeEmitEnd] (/home/alfredo/gits/xviz/node_modules/minipass/index.js:255:12)
(node:7927) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:7927) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
prebuild-install WARN install libvips-cpp.so.42: cannot open shared object file: No such file or directory
gyp info it worked if it ends with ok
gyp info using node-gyp@4.0.0
gyp info using node@10.16.0 | linux | x64
gyp info spawn /usr/bin/python2
gyp info spawn args [ '/home/alfredo/gits/xviz/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/home/alfredo/gits/xviz/node_modules/sharp/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/alfredo/gits/xviz/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/alfredo/.node-gyp/10.16.0/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
[... 10 lines cut ...]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/home/alfredo/gits/xviz/node_modules/sharp/build'
  TOUCH Release/obj.target/libvips-cpp.stamp
  CXX(target) Release/obj.target/sharp/src/common.o
In file included from ../vendor/include/glib-2.0/glib/galloca.h:32:0,
                 from ../vendor/include/glib-2.0/glib.h:30,
                 from ../vendor/include/glib-2.0/gobject/gbinding.h:28,
                 from ../vendor/include/glib-2.0/glib-object.h:23,
                 from ../vendor/include/vips/vips8:35,
                 from ../src/common.cc:25:
../vendor/include/glib-2.0/glib/gtypes.h:32:24: fatal error: glibconfig.h: No such file or directory
compilation terminated.
sharp.target.mk:126: recipe for target 'Release/obj.target/sharp/src/common.o' failed
make: Leaving directory '/home/alfredo/gits/xviz/node_modules/sharp/build'
make: *** [Release/obj.target/sharp/src/common.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/alfredo/gits/xviz/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.15.0-54-generic
gyp ERR! command "/usr/bin/node" "/home/alfredo/gits/xviz/node_modules/.bin/node-gyp" "rebuild"
gyp ERR! cwd /home/alfredo/gits/xviz/node_modules/sharp
gyp ERR! node -v v10.16.0

Qi2013, 2019-07-31:
I had a similar problem. It seems caused by a version of 'sharp' required by xviz. Sharp does not support node@12 before 0.22.1.
I solved the problem by changing the version of 'sharp' required by xviz/ros/ modules from 0.20.8 to 0.23.0, which is the same version required by the main xviz module.

My solution is inspired by the following discussion:
https://github.com/yarnpkg/yarn/issues/7265

DaHaiHuha, 2019-08-04:
I found the problem solved by down grading node to 10.15 which is listed in the package.xml, and the n other problems can be fixed one by another.

twojtasz, 2019-08-16:
@Qi2013 thanks for the details on the fix.