aurora-opensource/xviz #508: Support for Ellipse/Ellipsoid Geometry Primitive

dbruggner, 2019-08-05:
In https://avs.auto/#/xviz/protocol/schema/geometry-primitives there are different geometry primitives defined. But it seems that an ellipse/ellipsoid type is currently not supported in XVIZ. This type is very important to show the uncertainty of state estimates of objects in the scene which is mostly represented as a covariance matrix.
Could support for ellipse/ellipsoid geometry primitive be added?

jlisee, 2019-08-05:
Yes this is something I had wanted to add to XVIZ as well.   Would a flat ellipse be good enough for now?

dbruggner, 2019-08-06:
I think a flat ellipse will be a good start. Nevertheless, in the end only one ellipsoid type should be necessary.

twojtasz, 2019-08-16:
@dbruggner We have some new primitives we would like to add, I"ll be sure to add this to the list.

In the meantime, the best you can do would be to approximate it with a polygon.

You can paste the code below in the **frame** text window and see the ellipsoid.  Control+mouseMove to rotate the camera.

```// frame
const timestamp = 1001.0;

xvizBuilder
  .pose('/vehicle_pose')
  .timestamp(timestamp)
  .mapOrigin(-122.4, 37.8, 0)
  .orientation(0, 0, 0);

xvizBuilder
  .primitive('/object/shape')
  .polygon([
    [15, 5, 5],
    [14.755282581475768, 4.227457514062632, 5],
    [14.045084971874736, 3.530536869268817, 5.2],
    [12.938926261462367, 2.9774575140626314, 5.4],
    [11.545084971874736, 2.6223587092621163, 5.6],
    [10, 2.5, 5.8],
    [8.454915028125264, 2.622358709262116, 6],
    [7.061073738537635, 2.9774575140626314, 6.2],
    [5.954915028125264, 3.5305368692688166, 6.4],
    [5.244717418524233, 4.227457514062631, 6.6],
    [5, 5, 6.7],
    [5.244717418524232, 5.772542485937368, 6.6],
    [5.954915028125263, 6.4694631307311825, 6.4],
    [7.061073738537633, 7.022542485937368, 6.2],
    [8.454915028125262, 7.377641290737884, 6.0],
    [9.999999999999998, 7.5, 5.8],
    [11.545084971874736, 7.377641290737884, 5.6],
    [12.938926261462365, 7.022542485937369, 5.4],
    [14.045084971874736, 6.469463130731183, 5.2],
    [14.755282581475768, 5.772542485937369, 5]
  ])
  .id('object-1')
  .style({
    fill_color: '#08f7',
    height: 0.01
  }) ;  ```