aurora-opensource/xviz #525: xvizserver in live mode

DavidDiosdado, 2019-09-06:
Hi I am a soft engineer interested in this great project
I am trying to use the xviz server module in live mode with the streetscape getting-started example.  I have ran it in stream mode with some local data I generated myself using ./bin/babel-xvizserver -d ../../data/generated/... and the yarn start-streaming-local in the streetscape example but I can't figure out how to do it in live mode.

Could you please give me some guide about how to use the babel-xvizserver --live to transfer my live data from a running system source.

twojtasz, 2019-09-08:
@DavidDiosdado Could you describe more precisely what *live* mode means to you.

Unfortunately I do not believe our definition of live mode captures what most people may think.  Specifically, it relates to a system that is fundamentally missing some upfront information (like start time and end time.  Which mean we have to accept this information is missing, and the streetscape.gl application needs to manage data differently as well.   It will simply show the current *state* and not buffer or allow random access since that type of operation is not available with a *live* data stream.

There is a `yarn start-live-local` mode which makes some change in the streetscape.gl application to support the missing data described above.

From the XVIZ server side, in order for that to support a *live* data source you would have to augment it with a Provider that is able to server up the XVIZ data from your target system.

It can be done, we just need to clarify your environment so I can advise you ;)

DavidDiosdado, 2019-09-10:
My system is a python script that generates de json files in the XVIZ protocol in real time. I have tried to start the server using  ./bin/babel-xvizserver -d (dir where the script is storing the files) --live 
and then starting streetscape with yarn start-live-local but it didn't work.

I would like to know if this is possible to achieve using the live mode
Thanks for your attention

enginbaglayici, 2019-09-11:
Hello, I also want to generate .glb files and directly send it to Uberviewer to visualize data in real time. I can generate .glb files in real time using CAN data but I need some ideas to visualize this data with a frequency. If it possible, what would be the steps to achieve this? 
- Have should I edit log-from-file.js?
- Is only one frame.glb enough at a time for visualizing?
- Do I need a json file which has start time-end time infos (timingsFilePath)? (If I understood correct, I don't)
Thanks.

VijeetBenni, 2019-11-18:
I also need to visualize in runtime.

tried:
./bin/babel-xvizserver -d (dir where the script is storing the files) --live 1 --port 8081
&
yarn start-live-local

But getting below error:
get-started$ yarn start-live-local
**yarn run v1.17.3
warning package.json: No license field
$ webpack-dev-server --env.local --env.live --progress --hot --open
internal/modules/cjs/loader.js:638
    throw err;
    ^
Error: Cannot find module 'ocular-dev-tools/config/ocular.config'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/data/work/vbenni/workspace/LiDar/vizTool/uberAVS/streetscape/streetscape.gl/examples/webpack.config.local.js:31:17)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.**

enginbaglayici, 2019-11-18:
You can try the slack channel for your question. https://app.slack.com/client/TC45B521H/CGC3C8SMV/thread/CGC3C8SMV-1568123667.057900

VijeetBenni, 2019-11-18:
I solved the error.
But now after running **yarn start-live-local**

Not able to visualize anything on the streetscape.
Showing
**Missing data:/vehicle/acceleration**
**Missing data:/vehicle/velocity**
**Missing data:/vehicle/wheel_angle**

Not even images are displaying

heshiwen1989, 2020-03-09:
> I solved the error.
> But now after running **yarn start-live-local**
> 
> Not able to visualize anything on the streetscape.
> Showing
> **Missing data:/vehicle/acceleration**
> **Missing data:/vehicle/velocity**
> **Missing data:/vehicle/wheel_angle**
> 
> Not even images are displaying

Hello, Do you Solve This problem?

klu17, 2022-08-16:
@twojtasz I am running into a similar issue where I am unable to stream `/vehicle/acceleration`, `/vehicle/velocity`, and `/vehicle/wheel_angle`. Should the log file be modified? / Is there a solution to this common issue?

sriselva21-mbrdi, 2022-09-29:
Hi @DavidDiosdado ,

I am trying to run the command " ./modules/server/bin/babel-xvizserver -d data/generated/kitti/2011_09_26/2011_09_26_drive_0005_sync --port 8081" at xviz server and streetscape.gl/examples/get-started$ yarn start-streaming-local at streetscape.gl side but the server part is not running and i am using windows operating system

twojtasz, 2022-09-29:
Many have used the linux for windows to run. That may be an easier path.

sriselva21-mbrdi, 2022-09-29:
What is the correct command to run the xviz server on windows machine for yarn start-streaming-local at streetscape.gl side.

twojtasz, 2022-09-29:
Unfortunately I do not have access to a Windows machine currently

sriselva21-mbrdi, 2022-09-29:
When i try to install latest version of webpack dev server version 4, the streestcape.gl UI not able to load the data properly

DavidDiosdado, 2022-09-29:
I recommend you to use a linux based OS. It will make the work a lot easier. Install ubuntu in a small partition of your hard drive and follow the steps to run streetscape demo.

sriselva21-mbrdi, 2022-10-12:
I am trying to run the below commands to make xviz server in live mode
UI -  "yarn start-live or npm run start-live" in path of streetscape.gl-master\examples\get-started 
Server - "python xviz-master\python\examples\serve_scenarios.py
The server is passing the metadata and state_update json to the UI in the websocket request, if i pass only one state_update the metrics are displayed in the UI, If i pass multiple state_update with different timestamp the UI is not loading, can anyone help me or correct me if made any mistake.

**Metadata:**
{'type': 'xviz/metadata', 'data': {'version': '2.0.0', 'streams': {'/vehicle/velocity': {'category': 'TIME_SERIES', 'units': 'm/s^2', 'scalar_type': 'FLOAT'}, '/vehicle/acceleration': {'category': 'TIME_SERIES', 'units': 'm/s^2', 'scalar_type': 'FLOAT'}, '/vehicle/wheel_angle': {'category': 'TIME_SERIES', 'units': 'm/s^2', 'scalar_type': 'FLOAT'}}, 'ui_config': {'Metrics': {'name': 'Metrics', 'config': {'type': 'PANEL', 'children': [{'type': 'CONTAINER', 'children': [{'type': 'METRIC', 'streams': ['/vehicle/velocity'], 'title': '/vehicle/velocity', 'description': '/vehicle/velocity'}, {'type': 'METRIC', 'streams': ['/vehicle/acceleration'], 'title': '/vehicle/acceleration', 'description': '/vehicle/acceleration'}, {'type': 'METRIC', 'streams': ['/vehicle/wheel_angle'], 'title': '/vehicle/wheel_angle', 'description': '/vehicle/wheel_angle'}], 'name': 'Metrics Panel', 'layout': 'vertical'}], 'name': 'Metrics'}}}}}

**State Updates:**
{'type': 'xviz/state_update', 'data': {'update_type': 'snapshot', 'updates': [{'timestamp': 1317042372.349, 'time_series': [{'timestamp': 1317042372.349, 'streams': ['/vehicle/velocity', '/vehicle/acceleration', '/vehicle/wheel_angle'], 'values': {'doubles': [103.5147680214713, 99.6941896903947, 108.34400237629795]}}]}]}}
{'type': 'xviz/state_update', 'data': {'update_type': 'snapshot', 'updates': [{'timestamp': 1317042472.349, 'time_series': [{'timestamp': 1317042472.349, 'streams': ['/vehicle/velocity', '/vehicle/acceleration', '/vehicle/wheel_angle'], 'values': {'doubles': [203.5147680214713, 199.6941896903947, 208.34400237629794]}}]}]}}

Data with one timestamp value
[image]

Data with multiple timestamp value
[image]

I am using windows 10 operating system

tarichuo, 2024-02-21:
> I am trying to run the below commands to make xviz server in live mode UI - "yarn start-live or npm run start-live" in path of streetscape.gl-master\examples\get-started Server - "python [...]

Hello, I have same problem, have you solved it?

twojtasz, 2024-02-22:
Let me try this weekend

twojtasz, 2024-03-17:
I was just trying and realized this is a windows machine.  Unfortunately, i
do not have a windows machine to verify and fix this issue.

twojtasz, 2024-03-17:
I just tried searching for free windows VM's and there does appear to be a
way for me to get them!

Let me try to get this going that way.