aurora-opensource/xviz #526: Parse Annotations

eric1221bday, 2019-09-17:
Apologies if I missed something, but I have not been able to find any sign of annotation parsing code in the Xviz parser module. The annotation message type seems to be fully defined in the spec, but seems to not be captured in the timeslice parser used when parsing a a stream set. Is this functionality actually implemented?

twojtasz, 2019-09-20:
@eric1221bday you have found a gap in the implementation . We have planned for that but the use-case went away and never finished the implementation.

Could you provide a bit more details of your use case?  If there a strong story I can find a spot in the roadmap.

eric1221bday, 2019-09-25:
@twojtasz Sure, our use case is that we have situations where downstream modules might issue some kind of command or make a decision that we want reflected/visualized on some object that's published on another stream. Other than using annotations there are only a couple of other ways to achieve this, either my backend keeps state and I apply that state on the object while marshalling my xviz streams, or I have to publish the object alongside the command when that command is sent to me. Neither of which is desirable.

In the meantime though do you think it would be prudent to add a little warning log message in the parse function if it receives an xviz stream with annotations? That should at least reduce confusion, because currently that omission is effectively invisible to the user.