aurora-opensource/xviz #529: How to set up a live XVIZ Provider?

WesleyYue, 2019-09-29:
Hi, I'm trying to set up a toy example of playing back the kitti dataset "live" through rosbag play and I'm not sure how to proceed. I think I can start with rosbag-provider.js, and modify it to subscribe to a rostopic as a rosnode through something like roslibjs. However, I'm stuck and I have a few questions:
1. For the Provider, I'm not sure what I'm should be doing with getMessageIterator(), should I just implement a no-op and ignore the iterator parameter in xvizMessage? It doesn't make sense to iterate a live system that you have no playback control over. 
2. For streetscape.gl's XVIZStreamLoader, why is duration a parameter to pass through the url? Isn't this dictated by the xviz-server? Is the documentation just mistakenly copy and pasted from the XVIZWebsocketLoader base class?
3. In general, how does an XVIZSession relate to a XVIZProvider? ie. how does [this diagram](https://avs.auto/#/xviz/api-reference/xviz-server/overview) and [this diagram](https://avs.auto/#/xviz/api-reference/xviz-io/overview) relate? I'm a bit lost in the high level picture of how providers connect with handlers, sessions, and middleware.

Thanks

twojtasz, 2019-10-07:
Would love to see this working and hopefully get some contribution back.

1. Correct, since the iterator abstraction on a **live** state is largely ignorable, you could ignore it as necessary.  However a **live** system may have some connection state that I could see affecting iteration ... or not if you abstracted the reconnection logic internally.

2. XVIZStreamLoader is for loading an **off-line** log. As such it has **timestamp** & **duration** to allow you to pick arbitrary location within the log.  Imagine a long 1hr log where you only care to load the time from 23:48 + 30seconds.   The server (using the index file) would then only transfer the specified data

3. The key point is that a **Session** handles this `data -> middleware -> output`.   In the current setup, the ProviderSession is using a XVIZProvider to provide the data, pipes it to the XVIZMiddleware, which has a carefully structured Websocket middleware component to handle the output.

For your case, if you created a new **rosnode-provider**, you could override the classes the built-in **XVIZProviderFactory** will query for support of a request.

WesleyYue, 2019-12-07:
Thanks for your help. I ended up implementing this with the [python xviz implementation](https://github.com/cmpute/xviz.py) (which has since been merged to master) with the ROS python api.

closed 2019-12-07 by WesleyYue