aurora-opensource/xviz #530: Xviz server installation as a dependency not working

balag3, 2019-10-04:
I would like to include xviz server as a dependency in one our project but it fails during installation:
`npm i @xviz/server`  
output:  
`npm ERR! path ${Project_PATH}/node_modules/@xviz/server/bin/xvizserver   
npm ERR! code ENOENT   
npm ERR! errno -2   
npm ERR! syscall chmod   
npm ERR! enoent ENOENT: no such file or directory, chmod   
 '${Project_PATH}/node_modules/@xviz/server/bin/xvizserver'  
npm ERR! enoent This is related to npm not being able to find a file.  
npm ERR! enoent    `

I get the same error output if I manually add it to `package.json` and run npm install.

twojtasz, 2019-10-07:
I know the cause, the package.json is missing the entry to include the "bin" folder.
I will fix and publish a patch update later today.

twojtasz, 2019-12-06:
#556 should fix this.  I will release a new version once I address a few other bugs.

twojtasz, 2019-12-10:
@balag3 sorry for the long delay. If you try with`@xviz/server@1.0.3` and it still persists please re-open and ping me.  I tested on my mac and the *bin* directory was included when `npm install @xviz/server`

closed 2019-12-10 by twojtasz