aurora-opensource/xviz #532: Tracker: Python XVIZ Support

twojtasz, 2019-10-07:
Create XVIZ builders and related components to support XVIZ generation in Python.

## Phase 1 by Oct 16th
- [ ]  setup python module (see [tox+poetry](https://poetry.eustace.io/docs/faq/#is-tox-supported))
- [ ] setup basic python builder for pose, primitives
- [ ]  setup protobuf writing ([mypy-protobuf ](https://github.com/dropbox/mypy-protobuf) and [pyrobuf](https://github.com/appnexus/pyrobuf))
- [ ] verify viewing in streetscape.gl
- [ ] publish beta module to pypi

## Phase 2 by Nov 1st
- [ ] expand builder features to parity with JS
  - [ ] time-series
  - [ ] rest of primitives
- [ ] data generators & testing
- [ ] developer flow (CI/formatting/coverage/benchmarks)
- [ ] Support S3 xviz writer

cmpute, 2019-11-12:
[xviz.py](https://github.com/cmpute/xviz.py) if you are interested~

twojtasz, 2019-11-18:
@cmpute Would you be up for submitting this for inclusion in the main repo? We have some plans for this work and happy to sync on our plans and see if they are in alignment with your work.

cmpute, 2019-11-18:
@twojtasz No problem for me~ But how do you plan to include this in the main repo? I guess the main repo is only for Javascript version?

twojtasz, 2019-11-19:
@cmpute Awesome!  Let me figure that out, the key thing would be to be able to generate and run validation at the data level, rather than at a particular implementation level.  Over time, we can change the structure to have a top-level separation for language specific implementations.

The key goals would be to have language agnostic tooling and data at the top level (or some common directory).  Then we can setup the publishing pipelines and CI flows appropriately for each language implementation.

The focus is on XVIZ spec being the central theme and tooling around that hopefully benefitting from being centralized and enabling a lock-step feature development flow.

cmpute, 2019-11-19:
@twojtasz Sure, but I'm not actually very familiar with your dev flow and I guess my code structure is also quite different from yours. How do we integrate your code and mine?

twojtasz, 2019-11-19:
@cmpute The easiest would simply be to create a top-level `python` folder and then put your code in there effectively.  Eventually the JS specific parts could move to a `javascript` directory with other shared data at the top-level as necessary.

I will want to run this by some folks and maybe try a few things before committing.

As for the code, I have only started the Python work so I am still fleshing out my structure so I don't think that will be a problem.  We would want to reach agreement on the setup.  In particular we where setting this up to be future forward, so python 3.6+, using poetry, tox, black, mypy.  I am focusing on the builder, with some additional focus on improvements I want to make in areas I see as being weak in the current API (plus some more recent features added).

I did notice you have one package and I was trying to setup the packages to be independent (since not everyone needs server an builder necessarily), but that could happen later.

cmpute, 2019-11-19:
@twojtasz Feel free to include my code in your repository as long as you can credit me in your repo haha. I think I would change the license to a more flexible one.

In terms of the tool chain, my package actually only depends on little third-party packages in addition to builtin Python features, ie `websockets` for server, `protobuf` for protobuf support, `numpy` for common ndarray support. I think it's easy for you to add infrastructure like poetry, tox, etc, although I didn't have any experience with the packages you mentioned.

I tried to preserve the same package structure as yours, but still I changed to provide Python users with more Pythonic APIs. Also I want to note that in my project, the data storage is highly dependent on the protobuf protocol since I think protobuf provide strong type check. Basic server modules and various builders (almost all) have been already implemented, you can refer to the code to check the functionality coverage~

twojtasz, 2019-11-20:
@cmpute Absolutely your contribution would be credited.
Providing the pythonic API is the right call, and Protobuf is the direction we are heading as well.

I'm still collecting everything, and will be sure to add you to any PR's for review.

twojtasz, 2019-12-03:
@cmpute I am checking with our OSS team internally.  Right now the suggestion is that since you have a project under a GNU license, that we would need you to open a PR (and sign our Contributors License Agreement) and make note that you are contributing the code under the Apache license that XVIZ uses.

If you are interested in pursuing this let me we can keep it simple and just land this in a **python/<module>** directory.  We can keep the initial PR just code (and fix up module naming issues etc in follow-ups).

cmpute, 2019-12-03:
@twojtasz Sure, I'm fine with that

twojtasz, 2019-12-05:
@cmpute I'm going to be playing around on a separate branch in anticipation of this PR.  Let me know if you need any assistance.

I'm super eager and grateful for your work on this front!

cmpute, 2019-12-06:
@twojtasz Sorry for late of PR, I just created one and changed license to Apache, how can I sign the Contributors License Agreement?

twojtasz, 2019-12-06:
@cmpute looks awesome.  Thanks again for putting this up and doing it so fast!

twojtasz, 2019-12-06:
Just landed your PR!  Thank you very much for your contribution.

I plan for python related discussion to happen in our slack [#python channel. ](https://streetscapexviz.slack.com/archives/CR8LR6SCD).  Please join!

The first set of changes will just be some house-keeping.  Like changing the module name once I figure out what it should be to avoid conflict with you module already published.

cmpute, 2019-12-06:
@twojtasz If I recall right, I can hand the ownership of the package published to you and I can change my code to another package name, since your code will actually the "official implementation"~

cmpute, 2019-12-06:
And how can I join the slack channel? It seems to be an internal one since it requires XXX@tri.global email address

twojtasz, 2019-12-06:
@cmpute If you would be willing to transfer that would be great.  But lets figure out the details and see if it's necessary.

I also resent an invitation email.  I will try to fix the general signup and the **tri.global** issue.

cmpute, 2019-12-06:
@twojtasz Could you please send invitation again to this email zyxin@umich.edu, since the current one is my personal email address 😃 , sorry for the trouble

twojtasz, 2019-12-06:
Done. And I did remove the link that was tri.global specific.  And verified our docs have the general invitaion link.