aurora-opensource/xviz #535: running KITTI  example  failed

afei26579, 2019-10-16:
when I use : 
`./scripts/run-kitti-example.sh`
it work,
`# XVIZ Server started.`
`# Ctrl-c to terminate.`
`# xvizserver-log: Listening on port 8080`
but i open browser by http://localhost:8080/  ,it return:

"Upgrade Required"

what should I do ？
node version is 12.12
and apt update done.

balag3, 2019-10-16:
What happens here is that on http://localhost:8080 a websocket server is serving your data and the server tells you that it is missing an UPGRADE header from your browser request.
You need a client that is capable of opening a websocket connection. XVIZLiveLoader is one of them provided by xviz.
I personally use this extension: https://chrome.google.com/webstore/detail/websocket-test-client/fgponpodhbmadfljofbimhhlengambbn?hl=en  
Also what is missing from the description the address should be  ws://localhost:8080/

twojtasz, 2019-10-18:
@afei26579 This is an XVIZ *WebSocket* Server, not an HTTP Server.

To view this follow the *Quick Start* instructions on this page
https://github.com/uber/streetscape.gl

But replace `yarn start` with `yarn start-streaming-local` and it should work.

afei26579, 2019-10-28:
thanks lots . 

> @afei26579 This is an XVIZ _WebSocket_ Server, not an HTTP Server.
> 
> To view this follow the _Quick Start_ instructions on this page
> https://github.com/uber/streetscape.gl
> 
> But replace `yarn start` with `yarn start-streaming-local` and it should work.

thanks a lot.
a new question，i run sush : yarn start-ttreaming-local 
the Command log:
   
[image]
and browser :   
[image]

twojtasz, 2019-10-28:
@afei26579 Can you expand the error with the text "TypeError: can't convert null to object" . I have not seen this and am not sure where the issue is.  By expanding hopefully it will show the file and line.  If it says something like `bundle....` then it is not using source maps.  So try to click on the link on the right so I can try to get context on where this is happening.

twojtasz, 2019-12-10:
@afei26579 fixed in https://github.com/uber/xviz/pull/560

closed 2019-12-10 by twojtasz