aurora-opensource/xviz #537: Problem when serving generated KITTI dataset

eastlife, 2019-10-18:
Hello, I love this excellent project very much. When serving xviz of generated data, I cannot see the visualization in streetscape.gl. Actually I suppose the connection between xviz and streetscape.gl is established, because I can see the log "xvizserver-log: [> Connection] made" in xviz terminal, but there is no data from streetscape.gl (no lidar, black image, no velocity, etc). My dataset is generated from example KITTI dataset (a bunch of glb files). I am following this link:
https://github.com/uber/xviz/blob/master/docs/getting-started/running-the-example.md

twojtasz, 2019-10-28:
Not sure what could be wrong.  Does the browser console list any errors?  I will try from scratch as soon as I can and see if I can reproduce.
What platform are you running on?

eastlife, 2019-10-28:
There are many errors in the console. See the attached screenshot. And I am running on Ubuntu18.04
[image]
[image]

speculees, 2019-12-02:
This seems like it is linked to #540

twojtasz, 2019-12-10:
@eastlife Fixed in https://github.com/uber/xviz/pull/560

closed 2019-12-10 by twojtasz