aurora-opensource/xviz #538: Problem with ROS Converter

eastlife, 2019-10-18:
I've got another problem with ROS converter. Currently I can see the visualization of point cloud with the example ROSConfig json (config-1.json, config-2.json). However, when I add image data to ROSConfig, I cannot get the visualization of images as the example server. I am using the rosbag converted from the example KITTI dataset with kitti2bag, and I'm following the link below to install it:
https://github.com/uber/xviz/tree/master/examples/converters/ros

My ROSConfig is something like:
```
{
  "topicConfig": [
    {
      "topic": "/kitti/oxts/gps/fix",
      "type": "sensor_msgs/NavSatFix",
      "config": {
        "xvizStream": "/vehicle_pose",
        "imuTopic": "/kitti/oxts/imu"
      }
    },
    {
      "topic": "/kitti/camera_color_left/image_raw",
      "converter": "SensorCompressedImage",
      "config": {
        "xvizStream": "/vehicle/camera/center_front"
      }
    }
  ]
}
```
(the example from https://github.com/uber/xviz/blob/master/docs/api-reference/ros/ros-config.md)

twojtasz, 2019-10-28:
There is a bug in the ROS Image converter, that I have not fixed yet. so only a very small number of image formats are supported.

eastlife, 2019-10-28:
Currently what formats it supports? I am just running the example ROS bag(converted from example KITTI dataset)

YounHS, 2020-09-21:
Knock knock...
Has this problem been get solved??

I have something problem with my json(rosconfig) file...
When i start server, it doesn't show velocity, acceleration, camera...

I also set converter and config, but it doesn't work!!

plz, Give me some example file about config.json file

Thank U.