aurora-opensource/xviz #542: Converting custom data to XVIZ

patnolan33, 2019-11-04:
I am trying to convert a custom data set to XVIZ and I am having a hard time figuring out where to even begin from the documentation. I have a dataset that was collected from an open source simulator in the following format:

```bash
├── data
│   ├── gps
│   │   ├── timestamps.txt
│   │   ├── gps_points.txt
│   ├── camera
│   │   ├── timestamps.txt
│   │   ├── camera_id
│   │   │   ├── image_0.png
│   │   │   ├── image_1.png
.
.
.
│   │   │   ├── image_N.png
│   ├── lidar
│   │   ├── timestamps.txt
│   │   ├── lidar_id
│   │   │   ├── lidar_0.png
│   │   │   ├── lidar_1.png
.
.
.
│   │   │   ├── lidar_N.png
```

In each directory, there is a `timestamps.txt` file that consists of timestamps for each successive measurement from the simulation. For example, the first line of `gps->timestamps.txt` corresponds to the first line of `gps->gps_points.txt`. Similarly, the first line of `camera->timestamps.txt` corresponds to the first image (i.e.  `camera->camera_id->image_0.png`). For ease of development at the moment, all timestamps are synced.

I have looked through the code for converting to XVIZ from the KITTI format, and the documentation just doesn't seem to outline how those conversions can be done on a custom data format. The documentation [HERE](https://avs.auto/#/xviz/user-guide/generating-xviz) seems to indicate that pointing a `builder` instance to points loaded in JSON format would result in proper XVIZ data, but how is that then loaded via a file in `streetscape.gl`? How is this data saved to then be loaded later in some XVIZ viewer?

Is there any tutorial on taking custom data, converting it to XVIZ, and then loading it later into something like `streetscape.gl`?

twojtasz, 2019-12-02:
@patnolan33 KITTI is technically a custom data format, so KITTI should have all that is necessary to help you.

Having said that, if all of your timestamps are **synced** then you should be able to simply iterate by timestamp.  And for each data set create a **converter** that knows how to load the timestamps for that data and index to the data when given a timestamp. Inside your converter, you would have a builder that does the conversion to XVIZ.

For saving the data and loading it into streetscape.gl, you can reference https://avs.auto/#/xviz/getting-started/converting-to-xviz/viewing-the-generated-xviz?section=disable-cameras

This should allow you to generate the data and save to a directory.  Then the server will create a websocket server to host that data.   Then you start streetscape.gl in a **start-streaming-local** mode and it will look at localhost to find the data source.

jediofgever, 2020-04-02:
@patnolan33 Have you figured this out ? I al also trying to do the same now any feedback is appreciated

vbxz, 2020-09-18:
I am also very confused about this question.I do not know how to convert custom data to XVIZ.Do I need to write a converter by myself?And how to run converter to convert data?Looking forward to your reply.Thank you!

pai-qi, 2021-04-09:
> I am also very confused about this question.I do not know how to convert custom data to XVIZ.Do I need to write a converter by myself?And how to run converter to convert data?Looking forward to your reply.Thank you!

I have the same question, do you have the answer ?

patnolan33, 2021-04-09:
Sorry for the delayed response. I did not find a direct way to do this using XVIZ, and ended up writing my own custom converter for my purposes that uses a different visualization tool. I'm sorry I can't be of much help.

kvalev, 2021-04-09:
What we did in the past (but this might be an overkill) is to create a new provider and register it with xviz (see the native [ROSBagProvider](https://github.com/uber/xviz/blob/2ce7949b903e2e338ae3155cd7fbf21fca804505/modules/ros/src/providers/rosbag-provider.js#L57)). This was you will be able to convert your custom data format, either ahead-of-time or during runtime.