aurora-opensource/xviz #543: Not able to display images for type sensor_msgs/CompressedImage

hpanyda, 2019-11-06:
I am using ROS bag with compressed images. On Streescape.gl, I am able to visualize the point cloud but images from the cameras are not getting displayed. Please let me know if there is something that I am missing here.
The ROS bag that I have contains the following

    /camera_3/image_raw/compressed sensor_msgs/CompressedImage
    /camera_3/camera_info sensor_msgs/CameraInfo
    /camera_1/image_raw/compressed sensor_msgs/CompressedImage
    /camera_1/camera_info sensor_msgs/CameraInfo
    /camera_2/image_raw/compressed sensor_msgs/CompressedImage
    /camera_2/camera_info sensor_msgs/CameraInfo
    /camera_0/image_raw/compressed sensor_msgs/CompressedImage
    /camera_0/camera_info sensor_msgs/CameraInfo
    /velodyne_points sensor_msgs/PointCloud2

The rosConfig that I am using is as below

    { 
	   "topicConfig":[ 
		  { 
			 "topic":"/camera_3/image_raw/compressed",
			 "type":"sensor_msgs/CompressedImage"
		  },
		  { 
			 "topic":"/camera_3/camera_info",
			 "type":"sensor_msgs/CameraInfo"
		  },
		  { 
			 "topic":"/camera_1/image_raw/compressed",
			 "type":"sensor_msgs/CompressedImage"
		  },
		  { 
			 "topic":"/camera_1/camera_info",
			 "type":"sensor_msgs/CameraInfo"
		  },
		  { 
			 "topic":"/camera_2/image_raw/compressed",
			 "type":"sensor_msgs/CompressedImage"
		  },
		  { 
			 "topic":"/camera_2/camera_info",
			 "type":"sensor_msgs/CameraInfo"
		  },
		  { 
			 "topic":"/camera_0/image_raw/compressed",
			 "type":"sensor_msgs/CompressedImage"
		  },
		  { 
			 "topic":"/camera_0/camera_info",
			 "type":"sensor_msgs/CameraInfo"
		  },
		  { 
			 "topic":"/velodyne_points",
			 "type":"sensor_msgs/PointCloud2"
		  },
		  { 
			 "topic":"/velodyne_points",
			 "converter":"XVIZFakePose",
			 "config":{ 
				"xvizStream":"/vehicle_pose"
			 }
		  }
	   ]
	}

I also tried by adding explicit converter for all camera streams but the result is the same.

      "converter": "SensorCompressedImage",
      "config": {
        "xvizStream": "/camera_1/image_raw/compressed sensor_msgs/CompressedImage"
      }

Please let me know how we can display the compressed images on streetscape.gl. 

Regards,
Hitesh

hpanyda, 2019-11-09:
Anyone any inputs? 
Would like to decompress the compressedImage to jpeg/png and resize it with sharp to display.
Thanks & Regards,
Hitesh

hpanyda, 2019-11-11:
Hi,

I tried changing the ROS compressedImage to raw jpeg image and resized it with sharp but still it is not getting displayed on streetscape.gl. Point cloud is getting displayed properly.

Below are the changes done in xviz/modules/ros/src/messages/sensor-compressedimage-converter.js

	import Converter from './converter';
	import sharp from 'sharp';
	import cv from 'opencv4nodejs';

	export class SensorCompressedImage extends Converter {
	  constructor(config) {
		super(config);
	  }

	  static get name() {
		return 'SensorCompressedImage';
	  }

	  static get messageType() {
		return 'sensor_msgs/CompressedImage';
	  }

	  async convertMessage(frame, xvizBuilder) {
		const msgs = frame[this.topic];
		if (!msgs) {
		  return;
		}

		if (msgs.length) {
		  const {message} = msgs[msgs.length - 1];
		  const {format, data} = message;
		  
		  //decode compressed image to MAT
		  const croppedImage = cv.imdecode(data);
		  
		  //encode MAT back to .jpeg image
		  const buf =  cv.imencode('.jpeg', croppedImage);

		  // Converting it to standaed 400x300 for display
		  const width = 400;
		  const height = 300;
		 
		  const imgData =  await sharp(buf)
						.resize(width,height)
						.toFormat('jpeg')
						.toBuffer();
		  
		  xvizBuilder
		    .primitive(this.xvizStream)
		    .image(nodeBufferToTypedArray(imgData), 'jpeg')
		    .dimensions(width, height);
		}
	  }

	  getMetadata(xvizMetaBuilder) {
		const xb = xvizMetaBuilder;
		xb.stream(this.xvizStream)
		  .category('primitive')
		  .type('image');
	  }
	}

	function nodeBufferToTypedArray(buffer) {
	  // TODO - per docs we should just be able to call buffer.buffer, but there are issues
	  const typedArray = new Uint8Array(buffer);
	  return typedArray;
	}

Please note that swapping height and width to 400 and 300 respectively has no impact. I am able to write images to files and it is creating proper images.

Any inputs here will help.

Regards,
Hitesh

hpanyda, 2019-11-11:
Just to add, please note that format contains the following for front color camera. For now, I have also reduced the rosConfig json to have single xvizStream .

    bgr8; jpeg compressed bgr8

twojtasz, 2019-12-02:
@hpanyda Since they are compressed images, i would try to pass them through w/o sharp re-encoding.

Do you know if the data is present and correct?
If it is correct, have you configured the UI to be looking at the correct stream.  Specifically you can see where in **ros-custom** we define some custom UI mapping to the streams for cameras.

https://github.com/uber/xviz/blob/master/examples/converters/ros-custom/main.js#L37

If possible, a single message with an image for me to test would help me debug.

hpanyda, 2019-12-11:
Thanks and apologies for the delayed reply.
Please find attached the file with few messages.

Best regards,
Hitesh
[subset.bag.tar.gz](https://github.com/uber/xviz/files/3950167/subset.bag.tar.gz)

atefehkhoshnood, 2020-04-16:
how did you resolve this @hpanyda?

hpanyda, 2020-04-23:
Not resolved, got into other priorities.