aurora-opensource/xviz #549: Using multiple point clouds gives warning

mla2, 2019-11-21:
I'm trying to feed two lidar streams, one left and one right. I'm getting this on every update

`Point cloud for /lidar/left overwriting previous cloud`

I'm feeding alternating left and right, with incremental update. It seems to work, but the warning is a bit concerning. Am I doing something wrong, or is this bug, or a limitation?

It's a bit hard to merge them since they have a different transform and arrive with different time stamps. So sending them as two separate streams seems very convenient.

enricoschroeder, 2019-11-26:
Same "problem" here, I would also be interested in some insight on this topic.

twojtasz, 2019-12-02:
@mla2 @schroederen This is a legacy warning message that is inaccurate.
I will remove it since it is no longer a valid message.

PR #558 (2019-12-06): Remove warning on pointclouds that is no longer relevant
Fixes #549
commit 4981dbd9 (2019-12-06): Remove warning on pointclouds that is no longer relevant (#558)

Fixes #549
closed 2019-12-06 by twojtasz