aurora-opensource/xviz #550: Basic ROS Example - Stuttering and missing /vehicle_pose

Apollys, 2019-11-22:
I am following the instructions in the [Basic ROS Example](https://github.com/uber/xviz/tree/master/examples/converters/ros#example-1-viewing-a-single-topic), specifically using an `XVIZFakePose` to view the pointcloud from my own bag (not KITTI).

When run, there are quite a lot of `Every message requires a /vehicle_pose stream` messages output to the console.  In addition, I am seeing stuttering in the pointcloud visualization.  The KITTI example works much more smoothly and I get 5 such messages, but for my bag of the same length I get around 40 messages.

My bag is exactly 17 seconds long and has exactly 170 pointcloud messages, so I don't think missing messages are an issue.

What may be causing this behavior?

lance-plusai, 2019-11-27:
I have the same problem, there should be some problems with the code https://github.com/uber/streetscape.gl/blob/6a3549eba1ff122fe2135c87d576ac4dbe6ee255/modules/core/src/utils/transform.js#L34 .
 I simply fixed some, and everything is normal now: 
repase `  let parentPose = streams[streamName] && streams[streamName].target_pose;` with
`let parentPose = links[streamName] && links[streamName].target_pose;`

twojtasz, 2019-12-02:
The XVIZFakePose should be outputting the messages required on the stream `/vehicle_pose`.

If you could upload, a small sample of sequential xviz messages that would help to repo/debug.  Knowing that data may be private, it might be useful to debug and see what the timestamps are, and maybe output a random set of points at that same time you would output your point clouds. Then we could debug at the data level.

EdoardoSerri, 2021-02-05:
I have the same issue, but using /veihicle_pose. I have the same messages and the same composition as the KITTI ones, but the one from KITTI are very smooth while mine are very laggy. I thought the problem was about the sampling of my nav messages, but the difference was not so far. Is it a bug or am I missing something?