aurora-opensource/xviz #551: Polyline primitive is not displayed in case of too close GEOGRAPHIC coordinates.

balag3, 2019-11-23:
I would like to use the polyline primitive to mark the pose's covered route. 
In the metadata  I set the polyline's coordinates to GEOGRAPHIC, since what I get are live gps coordinates.
In every frame I push the pose's current position into a list  (Array:Point3D) and it is not displayed.
Playing with [playground](https://avs.auto/playground/?sample_streams=polyline) it is easy to reproduce the problem.
For example when the vertices are [[11.3909919,48.2958628,0], [11.3736340,48.284889,0]] the polyline is not displayed. When I change the at least the first decimal digit or the altitude, the polyline gets displayed.   
Examples:
 [[11.3xxxx, 48.2xxxx, 0], [11.3xxxx, 48.2xxxx, 0]] -> NOT DISPLAYED  
 [[11.3xxxx, 48.2xxxx, 0], [11.3xxxx, 48.**3**xxxx, 0]] -> DISPLAYED
 [[11.3xxxx, 48.2xxxx, 0], [11.**4**xxxx, 48.2xxxx,0]] -> DISPLAYED
 [[11.3xxxx, 48.2xxxx, 0], [11.3xxxx, 48.2xxxx, **1**]] -> DISPLAYED

twojtasz, 2019-12-02:
Thanks for the test data.

twojtasz, 2019-12-06:
For avs.auto/playground

metadata
`// metadata
xvizMetadataBuilder
  .startTime(1000)
  .endTime(1005)

  .stream('/vehicle_pose')
  .category('POSE')

  .stream('/objects/points')
  .type('POINT')
  .category('PRIMITIVE')
  .coordinate('GEOGRAPHIC')
  .streamStyle({radius_pixels: 5, point_color_mode: 'distance_to_vehicle'})

  .stream('/objects/lines')
  .type('POLYLINE')
  .category('PRIMITIVE')
  .coordinate('GEOGRAPHIC')
  .streamStyle({
        stroke_color: '#FFC043',
        stroke_width: 5,
        stroke_width_min_pixels: 1
      })`

Frame Message
`// frame
const timestamp = 1003.1;

xvizBuilder
  .pose('/vehicle_pose')
  .timestamp(timestamp)
  .mapOrigin(11.3908919,48.2959628, 0)
  .orientation(0, 0, 0)
xvizBuilder
   .primitive('/objects/points')
   .points([[11.3909919,48.2958628,0], [11.3736340,48.284889,0]])

// Original values - Nothing green shows up
xvizBuilder
   .primitive('/objects/lines')
   .polyline([[11.3909919,48.2958628,0], [11.3736340,48.284889,0]])
   .style({
    stroke_color: [0, 255, 0, 255]
  });
 
//Add z 0.1 and the orange color shows up
xvizBuilder
   .primitive('/objects/lines')
   .polyline([[11.3909919,48.2958628,0.1], [11.3736340,48.284889,0]])

// Changed 48.29 -> 48.39, and a red line shoots off the left
xvizBuilder
   .primitive('/objects/lines')
   .polyline([[11.3909919,48.3958628,0.1], [11.3736340,48.284889,0]])
   .style({
    stroke_color: [255, 0, 0, 255]
  });`

EliotCao, 2020-07-30:
I have a similar problem that polyline can't display correctly. I have test it in playground. 
the metedata is
`// metadata
xvizMetadataBuilder
  .startTime(1000)
  .endTime(1005)

  .stream('/vehicle_pose')
  .category('POSE')

  .stream('/prediction/trajectory')
  .category('PRIMITIVE')
  .type('POLYLINE')
  .coordinate('GEOGRAPHIC')
  .streamStyle({
    stroke_color: '#0f0'
  })` 

and the state_updata is 
`// frame
const timestamp = 1000.4;

xvizBuilder
  .pose('/vehicle_pose')
  .timestamp(timestamp)
  .mapOrigin(-83.046385,42.508557,0)
  .orientation(0, 0, 0)

xvizBuilder
  .primitive('/prediction/trajectory')
  .polyline([[-83.046385,42.508557,0], [-83.046512, 42.511854, 0], [-83.047093, 42.520714, 0], [-83.049987, 42.520722, 0], [-83.052827, 42.520669, 0]])
  .id('object-1')
  .style({stroke_width: 5})`

but after I changed the altitude value at intervals, it looks like correct.
`// frame
const timestamp = 1000.5;

xvizBuilder
  .pose('/vehicle_pose')
  .timestamp(timestamp)
  .mapOrigin(-83.046385,42.508557,0)
  .orientation(0, 0, 0)

xvizBuilder
  .primitive('/prediction/trajectory')
  .polyline([[-83.046385,42.508557,0], [-83.046512, 42.511854, 1], [-83.047093, 42.520714, 0], [-83.049987, 42.520722, 1], [-83.052827, 42.520669, 0]])
  .id('object-1')
  .style({stroke_width: 5})`

[image]

EliotCao, 2020-08-06:
By setting XVIZ Config "pathDistanceThreshold" to 0, the problems is solved.

twojtasz, 2020-08-06:
The threshold is intended to protect against rendering issues when
sequential points are very close

EliotCao, 2020-08-14:
> The threshold is intended to protect against rendering issues when sequential points are very close
> […](#)

Got it, thanks.